apache/nuttx #14006: [BUG] Unknown IOCTL in FUSB302 driver

TimJTi, 2024-10-09:
### Description / Steps to reproduce the issue

I am seeing unrecognized IOCTL error reports from the FUSB302 USBMISC driver recently. I suspect there's a recent change that has caused this and am looking for suggestions as to where to look to investigate.
It happens if (for example) I run a ping from nsh (albeit via a CDC/NCM usb connection so there is a relevance).

The IOCTL is 0x30a which I think is a file system IOCTL, FIONBIO, so not sure why this driver is being sent this?

This is the call stack. Can anyone advise?

```
fusb302_ioctl@0x2001b4d4 (/home/tim/designs/nuttx/nuttx/drivers/usbmisc/fusb302.c:1739)
file_vioctl@0x2003988c (/home/tim/designs/nuttx/nuttx/fs/vfs/fs_ioctl.c:68)
file_ioctl@0x20039b04 (/home/tim/designs/nuttx/nuttx/fs/vfs/fs_ioctl.c:225)
file_dup3@0x200397f0 (/home/tim/designs/nuttx/nuttx/fs/vfs/fs_dup2.c:150)
file_dup2@0x20039838 (/home/tim/designs/nuttx/nuttx/fs/vfs/fs_dup2.c:200)
files_duplist@0x20038f98 (/home/tim/designs/nuttx/nuttx/fs/inode/fs_files.c:723)
group_setuptaskfiles@0x2000e534 (/home/tim/designs/nuttx/nuttx/sched/group/group_setuptaskfiles.c:99)
nxtask_init@0x2000cfb4 (/home/tim/designs/nuttx/nuttx/sched/task/task_init.c:139)
nxtask_spawn_create@0x2000d6cc (/home/tim/designs/nuttx/nuttx/sched/task/task_spawn.c:108)
nxtask_spawn_exec@0x2000d6cc (/home/tim/designs/nuttx/nuttx/sched/task/task_spawn.c:224)
task_spawn@0x2000d6cc (/home/tim/designs/nuttx/nuttx/sched/task/task_spawn.c:321)
exec_builtin@0x200338ac (/home/tim/designs/nuttx/apps/builtin/exec_builtin.c:189)
nsh_builtin@0x200336a0 (/home/tim/designs/nuttx/apps/nshlib/nsh_builtin.c:105)
nsh_execute@0x20030928 (/home/tim/designs/nuttx/apps/nshlib/nsh_parse.c:650)
nsh_parse_command@0x20030928 (/home/tim/designs/nuttx/apps/nshlib/nsh_parse.c:2840)
nsh_parse@0x20030d74 (/home/tim/designs/nuttx/apps/nshlib/nsh_parse.c:2930)
nsh_session@0x2002a2c4 (/home/tim/designs/nuttx/apps/nshlib/nsh_session.c:245)
nsh_consolemain@0x2002a0f0 (/home/tim/designs/nuttx/apps/nshlib/nsh_consolemain.c:75)
nsh_main@0x2002a094 (/home/tim/designs/nuttx/apps/system/nsh/nsh_main.c:74)
nxtask_startup@0x2001e390 (/home/tim/designs/nuttx/nuttx/libs/libc/sched/task_startup.c:70)
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 22.04.5 LTS

### NuttX Version

Master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Drivers]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-10-12:
@xiaoxiang781216 any idea?

xiaoxiang781216, 2024-10-13:
FIONBIO is a general ioctl which may call from the common code, the driver should ignore any ioctl(include FIONBIO) which can't handle and return -ENOTTY without any other side effect.

TimJTi, 2024-10-14:
> FIONBIO is a general ioctl which may be called from the common code, the driver should ignore any ioctl(include FIONBIO) which can't handle and return -ENOTTY without any other side effect.

OK no worries - it does ignore it and return -ENOTTY as expected: I had turned on USB debug features which would be what caused this to show, didn't remember it from before.

closed 2024-10-14 by TimJTi