apache/nuttx #14010: [HELP] Making HTTPS calls using MBEDTLS

engdavidiogo, 2024-10-09:
### Description

I need help understanding what is wrong with this implementation attempt.

I am trying to make **HTTPS** calls in my project using **MBEDTLS**, and for that, I am using the default **_Wi-Fi_** configuration for the **_ESP32C6_**.

`$ ./tools/configure.sh esp32c6-devkitc:wifi`

After that, I enable the following configurations in the menu:

**_CONFIG_CRYPTO_MBEDTLS_**
**_CONFIG_MBEDTLS_APPS_**
**_CONFIG_MBEDTLS_APP_BENCHMARK_**
**_CONFIG_MBEDTLS_APP_SELFTEST_**

With this configuration, when trying to build, I get the following output:

[image]

Please let me know if I skipped any necessary steps for success.

### Verification

- [X] I have verified before submitting the report.

engdavidiogo, 2024-10-10:
Adding to what has already been said:

It seems like the issue is related to missing configurations in NuttX concerning the Espressif HAL for the ESP32C6.

As a validation, I enabled the MBEDTLS features using the following configuration for the STM32F746:

$ ./tools/configure.sh stm32f746g-disco:netnsh

[image]

I also applied the same configuration for the ESP32:

$ ./tools/configure.sh esp32-devkitc:wifi

[image]

@fdcavalcanti, @eren-terzioglu can you PTAL?

fdcavalcanti, 2024-10-11:
Check`nuttx/arch/risc-v/src/esp32c6/esp-hal-3rdparty/nuttx/esp32c6/include/sdkconfig.h`.
You will see some symbols are already defined there, maybe it will help you better understand what's going on.

pkarashchenko, 2024-10-13:
I was trying to deal with similar in the past in https://github.com/apache/nuttx/pull/11738
Seems like linking process is using `CFLAGS` with both app flags and flags from `Wireless.mk`

FelipeMdeO, 2024-10-14:
You are right @pkarashchenko , I did same thing than you and now we are able to compile.
@fdcavalcanti  can you check whether a did right process here: https://github.com/espressif/esp-hal-3rdparty/pull/5, please?

FelipeMdeO, 2024-10-14:
@engdavidiogo is everything ok in your side ?

engdavidiogo, 2024-10-14:
Yes @FelipeMdeO , I can reproduce the build without any errors.

@fdcavalcanti  I continue testing and have no apparent problems

FelipeMdeO, 2024-10-16:
@fdcavalcanti  We are able to update firmware using secure link, please lets update 3rdparty in nuttx master.

tmedicci, 2024-10-17:
> Reference

We are testing the changes internally. As soon as our CI finishes, we can merge it. Thanks!

tmedicci, 2024-10-18:
Hi @engdavidiogo and @FelipeMdeO ,

We merged it on `release/v5.1.c` branch:https://github.com/espressif/esp-hal-3rdparty/commit/26d4d1b57d54b63384b4e466b9e644ff0fe702b5

Can you please test it setting `ESP_HAL_3RDPARTY_VERSION=26d4d1b57d54b63384b4e466b9e644ff0fe702b5` before the `make` command to build NuttX? (we'd still need to update the hash for the HAL on NuttX)

engdavidiogo, 2024-10-22:
Hi @tmedicci thanks for the support with this!

Closing...

closed 2024-10-22 by engdavidiogo