apache/nuttx #14012: [HELP] nimBLE central (companion to examples/nimble) on a nrf52840-dk

maxikrie, 2024-10-09:
### Description

Hello,
I have successfully tested the examples/nimble app on a nrf52840-dk acting as a BLE peripheral. I am connecting to the board with a smartphone and the nrf connect app. I am trying to create a BLE central on a second nrf52840-dk, which can connect to the peripheral.

So far, I have examined and copied from [esp example](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/ble_multi_conn/ble_multi_conn_cent/main/main.c) and [mynewt-nimble example](https://github.com/apache/mynewt-nimble/blob/master/apps/blecent/src/main.c). However, when I perform a BLE scan for peripherals, `ble_gap_disc` or `ble_gap_ext_disc` (with `MYNEWT_VAL(BLE_EXT_ADV)`) fail with error code 513 hinting to a timeout in `ble_hs_hci_cmd_tx`.

I am happy to provide my code if necessary, but wanted to reach out first if anybody accomplished to get a BLE central to scan for peripherals?

Thanks! 

### Verification

- [X] I have verified before submitting the report.

maxikrie, 2024-10-12:
@raiden00pl I see that you added the nrf52840-dk:sdc_nimble config. Thank you for this, it helped me a lot to get started.

Does the above issue mean anything to you and/or have you managed to get a central as a companion to your example running? From the examples I referenced it shouldn't be hard to get at least the scanning running, but I am stuck with the issue I mentioned.

acassis, 2024-10-12:
@maxikrie I think Nimble integration on NuttX is something that many people are interested on. Could you please include a Documentation/ maybe it could be included like a Guide: https://nuttx.apache.org/docs/latest/guides/index.html

@pkarashchenko when you used BLE with ESP32xx did you used Nimble stack as well?

maxikrie, 2024-10-12:
@acassis So far, I only have the existing nrf52840-dk:sdc_nimble example running, which employs the Nimble stack with the Nordic Softdevice controller to create a BLE peripheral. My goal is to also have a BLE central either on another nrf52840-dk or some ESP32 board with Nimble. As I understand @pkarashchenko may already has a ESP32 configuration with Nimble (not sure if peripheral or central), which he will provide soon.

I am happy to provide a guide if I have it running.

raiden00pl, 2024-10-13:
hi, I haven't had a chance to play with nimble as a BLE central role. The problem is rather on the nimble side (maybe something missing in nimble configuration). Last time I looked, `nrf52840-dk/sdc` was working as a central with NuttX BLE host implementation. 
I think we should have nimble example with a BLE central role in `nuttx-apps` and port more examples from nimble (https://github.com/apache/mynewt-nimble/tree/master/apps)

maxikrie, 2024-10-14:
I have found the issue, which was a wrong configuration. I did set the nimble role to central, but I was missing that the softdevice role also needs to be set (otherwise it links the wrong nrxlib library).

The following needs to be set in the defconfig;
`
CONFIG_NIMBLE_ROLE_CENTRAL=y
CONFIG_NRF52_SDC_CENTRAL=y
CONFIG_NRF52_SDC_PERIPHERAL_COUNT=0
`
I managed to connect between a nrf52840-dk running nuttx with nimble as a central, and a ESP32 running the esp-idf with nimble as peripheral. Once I have everything in nuttx with nimble only, I will create a pull request.

closed 2024-10-14 by maxikrie

raiden00pl, 2024-10-16:
@acassis there is no nimble central example application in apps now, so a new board config doesn't make sense here :)

acassis, 2024-10-16:
> @acassis there is no nimble central example application in apps now, so a new board config doesn't make sense here :)

So the first step is creating a bluetooth central example application first. @maxikrie do you have some basic example that could be used to demonstrate it?

maxikrie, 2024-10-16:
Yes, I just got it working. I will provide the source with a config for the nrf52840-dk soon, probably next week.

maxikrie, 2024-10-30:
@acassis @raiden00pl
I have created a pull request on the nuttx-apps repository, which adds nimble_bleprph and nimble_blecent working in companion. These are the pull requests for apps (https://github.com/apache/nuttx-apps/pull/2807) and board configs (https://github.com/apache/nuttx/pull/14566)