apache/nuttx #14062: [HELP] NuttX release flow documentation

fdcavalcanti, 2024-10-10:
### Description

Hi all!

Just so I understand what's going on.

Which standard does NuttX follow during release? Do we have that written somewhere? I checked the [documentation](https://nuttx.apache.org/docs/latest/contributing/workflow.html) and it does not say.

It's not clear to me what are target release dates, how and when we sync release/master and how testing works during this process (does the CI run automatically for release branches on both `apps` and `nuttx` for instance).

Am I missing something on the documentation or maybe there is another forum?
This could be a moment to gather some suggestions and improve the docs.

Would appreciate if someone could enlighten me.

Thanks!

### Verification

- [X] I have verified before submitting the report.

lupyuen, 2024-10-11:
Thanks @fdcavalcanti for the great question! Yep the NuttX Release Process really needs to be documented, here's what I learnt so far:

- We target to have a NuttX Release __Every 3 Months__

- NuttX Release Process begins with the __Staging of the Source Code__ for `nuttx` repo [(releases/12.7)](https://github.com/apache/nuttx/tree/releases/12.7) and`nuttx-apps` repo [(releases/12.7)](https://github.com/apache/nuttx-apps/tree/releases/12.7)

- __CI Workflow__ will rebuild `nuttx` repo [(releases/12.7)](https://github.com/apache/nuttx/tree/releases/12.7) with the matching version of `nuttx-apps` repo [(releases/12.7)](https://github.com/apache/nuttx-apps/tree/releases/12.7)

- Which happens because of this [__CI Workflow Logic__](https://github.com/apache/nuttx/blob/master/.github/workflows/build.yml#L36-L110)

- [Issues with the Staged Source Code](https://github.com/apache/nuttx/pull/13785#issuecomment-2387967107) (fail to build / fail to run) will require fixes to be Backported from `master` branch to `releases/12.7` [(Like this)](https://github.com/apache/nuttx/pull/13788)

- The Release Source Code is published for [Validation and Voting](https://lists.apache.org/thread/gd5wwgkdv50o33h1h5os8js8sch1y4z9)

- __Validate NuttX Release__ is the final step of the Release Process, which checks that everything works OK on Real Hardware. [Check out the details here](https://lupyuen.github.io/articles/pr#appendix-validate-nuttx-release)

- __If the Vote Fails:__ We backport the required fixes, and bump up the release from RC0 to RC1 for validation and voting

Hi @acassis: Anything I missed?

jerpelea, 2024-10-11:
@lupyuen here is the release process 
https://gist.github.com/jerpelea/67895da8797f040ba3b113280fdcd57b

acassis, 2024-10-11:
@lupyuen I think you listed all the steps correctly. Maybe we can use these topics you listed together with the documentation from @jerpelea to create an official documentation. What do you think?

acassis, 2024-10-31:
@fdcavalcanti I think we can keep it opened, as it it an issue that needs to be fixed.

closed 2024-11-04 by xiaoxiang781216