apache/nuttx #14151: [BUG] Error: Image size (0x5000004c) + trailer (0x630) exceeds requested size 0x150000

metarutaiga, 2024-10-12:
### Description / Steps to reproduce the issue

I found it miss AT > ROM

```diff
diff --git a/boards/risc-v/esp32c3/common/scripts/esp32c3_sections.ld b/boards/risc-v/esp32c3/common/scripts/esp32c3_sections.ld
index cc2661327c..485f24a7f8 100644
--- a/boards/risc-v/esp32c3/common/scripts/esp32c3_sections.ld
+++ b/boards/risc-v/esp32c3/common/scripts/esp32c3_sections.ld
@@ -451,7 +451,7 @@ SECTIONS
   {
     . = ALIGN(4);
     *(.rtc.literal .rtc.text)
-  } >rtc_iram_seg
+  } >rtc_iram_seg AT > ROM
 
   /* RTC BSS section. */
 
```

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS

### NuttX Version

12.7.0

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-10-12:
@fdcavalcanti @tmedicci @eren-terzioglu please take a look ^

fdcavalcanti, 2024-10-13:
Thanks for reporting. This was addressed for C6 already on #13795. We'll check on C3.

fdcavalcanti, 2024-10-21:
Hi @metarutaiga please give us details on how to reproduce the issue.
Which defconfig you used, steps, all that.

metarutaiga, 2024-10-21:
```
./tools/configure.sh esp32c3-generic:sta_softap
make
```

and some codes must be patch like
```
diff --git a/netutils/dhcpd/dhcpd.c b/netutils/dhcpd/dhcpd.c
index 2f8f280d5..d53a33c38 100644
--- a/netutils/dhcpd/dhcpd.c
+++ b/netutils/dhcpd/dhcpd.c
@@ -516,7 +516,7 @@ static in_addr_t dhcpd_allocipaddr(void)
       lease = dhcpd_findbyipaddr(ipaddr);
       if ((!lease || dhcpd_leaseexpired(lease)))
         {
-#ifdef CONFIG_CPP_HAVE_WARNING
+#ifdef CONFIG_CPP_HAVE_WARNING_PASS
 #  warning "FIXME: Should check if anything responds to an ARP request or ping"
 #  warning "       to verify that there is no other user of this IP address"
 #endif
```

fdcavalcanti, 2024-10-30:
@metarutaiga please check if the modifications on #14563 solve your problem.

closed 2024-11-03 by metarutaiga