apache/nuttx #14167: [BUG] new warnings when building sim/smp

raiden00pl, 2024-10-12:
### Description / Steps to reproduce the issue

build with: gcc (GCC) 14.1.1 20240522

```
[raiden00:~/git/RTOS/nuttx/nuttx]$ make -j                                                                                                                                                                                                                            
Create version.h
LN: platform/board to /home/raiden00/git/RTOS/nuttx/apps/platform/dummy
Register: hello
Register: taskset
Register: getprime
Register: ostest
Register: nsh
Register: smp
Register: sh
CP:  /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/config.h
CP:  /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/fs/hostfs.h
machine/arch_atomic.c: In function ‘__atomic_fetch_add_2’:
machine/arch_atomic.c:120:22: warning: infinite recursion detected [-Winfinite-recursion]
  120 |   type weak_function __atomic_fetch_add_##n (FAR volatile void *ptr,   \
      |                      ^~~~~~~~~~~~~~~~~~~
machine/arch_atomic.c:450:1: note: in expansion of macro ‘FETCH_ADD’
  450 | FETCH_ADD(2, uint16_t)
      | ^~~~~~~~~
In file included from /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/atomic.h:84,
                 from /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/fs/fs.h:40,
                 from /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/sched.h:47,
                 from /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/arch.h:89,
                 from /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:38,
                 from machine/arch_atomic.c:31:
In function ‘spin_lock_wo_note’,
    inlined from ‘spin_lock_irqsave_wo_note’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:531:7,
    inlined from ‘spin_lock_irqsave’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:586:11,
    inlined from ‘__atomic_fetch_add_2’ at machine/arch_atomic.c:450:1:
/home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:243:5: note: recursive call
  243 |     atomic_fetch_add((FAR atomic_ushort *)&lock->tickets.next, 1);
      |     ^~~~~~~~~~~~~~~~
In function ‘spin_lock_wo_note’,
    inlined from ‘spin_lock_irqsave_wo_note’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:535:7,
    inlined from ‘spin_lock_irqsave’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:586:11,
    inlined from ‘__atomic_fetch_add_2’ at machine/arch_atomic.c:450:1:
/home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:243:5: note: recursive call
  243 |     atomic_fetch_add((FAR atomic_ushort *)&lock->tickets.next, 1);
      |     ^~~~~~~~~~~~~~~~
machine/arch_atomic.c: In function ‘__atomic_load_2’:
machine/arch_atomic.c:51:22: warning: infinite recursion detected [-Winfinite-recursion]
   51 |   type weak_function __atomic_load_##n (FAR const volatile void *ptr, \
      |                      ^~~~~~~~~~~~~~
machine/arch_atomic.c:354:1: note: in expansion of macro ‘LOAD’
  354 | LOAD(2, uint16_t)
      | ^~~~
In function ‘spin_lock_wo_note’,
    inlined from ‘spin_lock_irqsave_wo_note’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:531:7,
    inlined from ‘spin_lock_irqsave’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:586:11,
    inlined from ‘__atomic_load_2’ at machine/arch_atomic.c:354:1:
/home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:244:10: note: recursive call
  244 |   while (atomic_load((FAR atomic_ushort *)&lock->tickets.owner) != ticket)
      |          ^~~~~~~~~~~
In function ‘spin_lock_wo_note’,
    inlined from ‘spin_lock_irqsave_wo_note’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:535:7,
    inlined from ‘spin_lock_irqsave’ at /home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:586:11,
    inlined from ‘__atomic_load_2’ at machine/arch_atomic.c:354:1:
/home/raiden00/git/RTOS/nuttx/nuttx/include/nuttx/spinlock.h:244:10: note: recursive call
  244 |   while (atomic_load((FAR atomic_ushort *)&lock->tickets.owner) != ticket)
      |          ^~~~~~~~~~~
LD:  nuttx  

```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Kernel]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-12:
it's strange that ci can't catch this compiler warning. @crafcat7 please take a look.

anchao, 2024-10-12:
@raiden00pl please help to review this PR https://github.com/apache/nuttx/pull/14198

closed 2024-10-13 by anchao