apache/nuttx #14257: [HELP] Boot multi-core on FVP A510, but the code still run "while (up_testset(lock) == SP_LOCKED)"  loop in spin_lock()

lzc1001, 2024-10-14:
### Description

Hi everyone!
I want to boot multi-core on FVP A510. After cpu1 run to WFI instruction, cpu0 run to "while (up_testset(lock) == SP_LOCKED)"  loop. This flow is: sched_unlock() (in nx_start()) -> nxsched_merge_pending() -> nxsched_add_readytorun() -> "DEBUGVERIFY(up_cpu_pause(cpu));" -> "spin_lock(&g_cpu_paused[cpu]);" -> spin_lock() -> "while (up_testset(lock) == SP_LOCKED)" in spin_lock(), then the code is still run in this loop. the input parameter "cpu" of up_cpu_pause() is 1 because of the task in “btcb” belong to cpu1 but the code run on cpu0 currently, and the cpu1 is in idle loop. So, it seem that cpu0 is waiting cpu1 to end its task but cpu1 task is still running. I don't know how to solve this issue and I don't konw the task scheduler how to work correctly in this case  
Thank you very much!
[image]
[image]
[image]
[image]
[image]

### Verification

- [X] I have verified before submitting the report.

anchao, 2024-10-15:
@hujun260

hujun260, 2024-10-15:
@lzc1001 
You are using an older version of the code. In the latest community implementation, the synchronous pause function has been removed. It is recommended that you synchronize with the latest community code and try again.

Based on the problem you described, the issue seems to be that CPU1 is unable to respond to interrupts.

hujun260, 2024-10-15:
The old version can also run well,  but the new version will have significantly improved performance, especially in SMP.

lzc1001, 2024-10-16:
@hujun260 @anchao OK! Thank you very much!
I solve this issues by define "CONFIG_ARCH_SET_VMPIDR_EL2" in configuration file. The GIC was initilized incorrectly because the value of MPIDR_EL1 is incorrect in EL1. And it would casue an incorret cpuid and initilized GIC redistributor incorrectly. So, When I set VMPIDR_EL2 in EL2, this isuse is solved, hhh✌

lzc1001, 2024-10-16:
Hi, I have another question for function "int arm64_gic_send_sgi()". I find that read vaule of GICR_IGROUP0 is 0x0 in non-secure EL1, but the value is 0xDFFF00FF in secure and the value(0xDFFF00FF) is set by ARM TF-A in EL3. So, Is the "if " judgement condition reasonable when the core run in non-secure EL1?
[image]
Those two pictures are from Learn the architecture - Generic Interrupt Controller v3 and v4, Overview
[image]
[image]

hujun260, 2024-10-17:
There are modifications to arm64_gic_send_sgi in the latest code. Please check it out first.

lzc1001, 2024-10-17:
@hujun260 I download the latest code(commit 204871513486e940fbb29a9a524de9b9e063d3e0 (HEAD -> master, origin/master, origin/HEAD)), but the function "arm64_gic_send_sgi" is same as my location code(commit 7a865b247969a9dbc06eddcaf58c59a14dd05d6d (HEAD -> releases/12.6, tag: nuttx-12.6.0-RC1, origin/releases/12.6)), I'm just curious about this issue. hhh

closed 2024-10-21 by lzc1001