apache/nuttx #14260: [BUG] <stdio.h include nuttx/fs/fs.h nuttx/lib/lib.h Leading to duplicate definitions with third-party libraries>

anjiahao1, 2024-10-14:
### Description / Steps to reproduce the issue

I have been using libarchive in nuttx, and everything was fine until stdio.h included the nuttx header file. There was a conflict between the header file in nuttx and libarchive. I think I should modify `stdio.h` so that stdio.h does not include any `nuttx/*.h`. I tried to delete `nuttx/*.h `but there were many compilation errors.

```
libarchive/libarchive/archive_read_open_filename.c:89:17: error: conflicting types for ‘file_read’; have ‘ssize_t(struct archive *, void *, const void **)’ {aka ‘long int(struct archive *, void *, const void **)’}
   89 | static ssize_t  file_read(struct archive *, void *, const void **buff);
      |                 ^~~~~~~~~
/home/ajh/work/archivefs/nuttx/include/nuttx/fs/fs.h:1426:9: note: previous declaration of ‘file_read’ with type ‘ssize_t(struct file *, void *, size_t)’ {aka ‘long int(struct file *, void *, long unsigned int)’}
 1426 | ssize_t file_read(FAR struct file *filep, FAR void *buf, size_t nbytes);
      |         ^~~~~~~~~
libarchive/libarchive/archive_read_open_filename.c:90:17: error: conflicting types for ‘file_seek’; have ‘int64_t(struct archive *, void *, int64_t,  int)’ {aka ‘long int(struct archive *, void *, long int,  int)’}
   90 | static int64_t  file_seek(struct archive *, void *, int64_t request, int);
      |                 ^~~~~~~~~
/
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

ubunt24,04

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Api], [Area: Build System]

### Verification

- [X] I have verified before submitting the report.

anchao, 2024-10-15:
@Donny9 @guidingli

Sarthak-Pandey, 2026-10-13:
Hi @xiaoxiang781216 , I’m interested in contributing to this issue and would be happy to investigate the header dependency conflicts and work toward a solution.

I noticed that the issue is already assigned, so I wanted to check whether there might be an opportunity to collaborate with the current assignee or take up a related part of the investigation.

I’d be glad to contribute wherever it would be most helpful to the project. Thank you for your time and guidance!

xiaoxiang781216, 2026-10-14:
sure, no problems. please note this patch: https://github.com/apache/nuttx/pull/20565 which finish the half work(kernel mode). you can continue on the flat/protected mode. @casaroli

Sarthak-Pandey, 2026-10-17:
Hi @xiaoxiang781216 ,

Thank you for the pointer to PR #20565! That was a huge step forward in cleaning up `<stdio.h>`.

I am looking to continue this work for **Flat and Protected modes** so that third-party libraries (like `libarchive`, which collide on internal functions like `file_read`) can compile cleanly across all MCU targets.

Before opening a PR, I wanted to get your thoughts on the best approach forward:

### Approach 1: Full Decoupling (Mirroring #20565 across Flat/Protected)
* Completely remove `#include <nuttx/fs/fs.h>` from `<stdio.h>` for all builds.
* Audit and patch all files across `arch/`, `drivers/`, `fs/`, and `net/` that relied on `<stdio.h>` to transitively pull `fs.h` (preliminary audit shows ~75 files).
* *Advantage:* 100% clean POSIX `<stdio.h>` across the entire repository with zero leaked kernel types or function names.
* *Trade-off:* Touches ~75 files in `nuttx` plus potential companion updates in `nuttx-apps`.

### Approach 2: Two-Step Phased PRs
* **PR 1:** Add the explicit `#include <nuttx/fs/fs.h>` to the affected drivers and architecture files *first* (safe, non-breaking).
* **PR 2:** Once PR 1 and any `nuttx-apps` dependencies are merged, remove `fs.h` from `<stdio.h>`.
* *Advantage:* Easier to review, prevents breaking CI, and ensures clean git bisectability.

### Approach 3: Symbol Guarding in `fs.h`
* Keep `fs.h` in `<stdio.h>` for Flat mode for now, but guard conflicting internal function prototypes (e.g., `file_read`, `file_seek`, `file_ioctl`) behind internal guards so user-space and third-party libraries don't see them.
* *Trade-off:* Does not solve collisions with `struct file` or `struct inode`, only function names.

Do you prefer we go with **Approach 1 (or 2)** to permanently solve the indirect dependency issue, or is there another preferred direction for Flat mode?

Thanks!

xiaoxiang781216, 2026-10-17:
@Sarthak-Pandey it's better to apply the approach 2, thanks.

closed 2026-10-26 by xiaoxiang781216