apache/nuttx #14263: [HELP] ESP32-C3 Lower memory on 12.7

metarutaiga, 2024-10-14:
### Description

I upgrade to 12.7 but it show lower memory then 12.6

```
NuttShell (NSH) NuttX-12.6.0
nsh> free
                 total       used       free    maxused    maxfree  nused  nfree
      Umem:     316092     163716     152376     223768      96304    337      7
```

```
NuttShell (NSH) NuttX-12.7.0
nsh> free
                 total       used       free    maxused    maxfree  nused  nfree
      Umem:     282756     136676     146080     137944     145936    230      5
```

### Verification

- [X] I have verified before submitting the report.

fdcavalcanti, 2024-10-14:
Thanks for reporting.
Is your apps directory also on the correct release?
Please share which configuration and board you are using, then I can verify.

metarutaiga, 2024-10-14:
## Build
```
./tools/configure.sh esp32c3-generic:sta_softap
make
```

## 12.6
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd89a0,len:0x1458
load:0x403c7000,len:0x21cc
load:0x403d0000,len:0x22ec
entry 0x403d2298
[esp32c3] [INF] *** Booting MCUboot build v2.1.0 ***
[esp32c3] [INF] [boot] chip revision: v0.4
[esp32c3] [INF] [boot.esp32c3] SPI Speed      : 40MHz
[esp32c3] [INF] [boot.esp32c3] SPI Mode       : DIO
[esp32c3] [INF] [boot.esp32c3] SPI Flash Size : 4MB
[esp32c3] [INF] [boot] Enabling RNG early entropy source...
[esp32c3] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32c3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Boot source: primary slot
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Disabling RNG early entropy source...
[esp32c3] [INF] br_image_off = 0x10000
[esp32c3] [INF] ih_hdr_size = 0x20
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] DRAM segment: start=0x37199, size=0x1c30, vaddr=0x3fc85990
[esp32c3] [INF] IRAM segment: start=0x31809, size=0x5990, vaddr=0x40380000
[esp32c3] [INF] start=0x4038325e

NuttShell (NSH) NuttX-12.6.0
nsh> free
                 total       used       free    maxused    maxfree  nused  nfree
      Umem:     321756      51212     270544      51568     270512    103      2
nsh> 
```

## 12.7
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382088
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd89a0,len:0x1458
load:0x403c7000,len:0x21cc
load:0x403d0000,len:0x22ec
entry 0x403d2298
[esp32c3] [INF] *** Booting MCUboot build v2.1.0 ***
[esp32c3] [INF] [boot] chip revision: v0.4
[esp32c3] [INF] [boot.esp32c3] SPI Speed      : 40MHz
[esp32c3] [INF] [boot.esp32c3] SPI Mode       : DIO
[esp32c3] [INF] [boot.esp32c3] SPI Flash Size : 4MB
[esp32c3] [INF] [boot] Enabling RNG early entropy source...
[esp32c3] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32c3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Boot source: primary slot
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Disabling RNG early entropy source...
[esp32c3] [INF] br_image_off = 0x10000
[esp32c3] [INF] ih_hdr_size = 0x20
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] DRAM segment: start=0x19798, size=0x19c0, vaddr=0x3fc89760
[esp32c3] [INF] IRAM segment: start=0x10040, size=0x9758, vaddr=0x40380000
[esp32c3] [INF] start=0x4038501a

NuttShell (NSH) NuttX-12.7.0
nsh> free
                 total       used       free    maxused    maxfree  nused  nfree
      Umem:     287668      54316     233352      54672     233320    109      2
nsh> 
```

metarutaiga, 2024-10-15:
### 12.7 with ESP-HAL-8e8e62cf6a7ae8a9659b91706024cab38af29118

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd89a0,len:0x1458
load:0x403c7000,len:0x21cc
load:0x403d0000,len:0x22ec
entry 0x403d2298
[esp32c3] [INF] *** Booting MCUboot build v2.1.0 ***
[esp32c3] [INF] [boot] chip revision: v0.4
[esp32c3] [INF] [boot.esp32c3] SPI Speed      : 40MHz
[esp32c3] [INF] [boot.esp32c3] SPI Mode       : DIO
[esp32c3] [INF] [boot.esp32c3] SPI Flash Size : 4MB
[esp32c3] [INF] [boot] Enabling RNG early entropy source...
[esp32c3] [INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[esp32c3] [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[esp32c3] [INF] Boot source: primary slot
[esp32c3] [INF] Image index: 0, Swap type: none
[esp32c3] [INF] Disabling RNG early entropy source...
[esp32c3] [INF] br_image_off = 0x10000
[esp32c3] [INF] ih_hdr_size = 0x20
[esp32c3] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32c3] [INF] DRAM segment: start=0x159a8, size=0x15c8, vaddr=0x3fc85970
[esp32c3] [INF] IRAM segment: start=0x10040, size=0x5968, vaddr=0x40380000
[esp32c3] [INF] start=0x40383232

NuttShell (NSH) NuttX-12.7.0-RC0
nsh> free
                 total       used       free    maxused    maxfree  nused  nfree
      Umem:     304380      51220     253160      51576     253128    103      2
nsh>
```

eren-terzioglu, 2024-10-15:
Thanks for the details @metarutaiga. We will look into that.

metarutaiga, 2024-10-19:
Show free memory after make

```diff
diff --git a/tools/espressif/Config.mk b/tools/espressif/Config.mk
index 4e76bd1d0c..ab6e659c8b 100644
--- a/tools/espressif/Config.mk
+++ b/tools/espressif/Config.mk
@@ -136,6 +136,10 @@ define MKIMAGE
        imgtool sign $(IMGTOOL_SIGN_ARGS) nuttx.hex nuttx.bin
        $(Q) echo nuttx.bin >> nuttx.manifest
        $(Q) echo "Generated: nuttx.bin (MCUboot compatible)"
+       $(Q) git log -1 --pretty=" * nuttx:%cs (%h)"
+       $(Q) git -C arch/risc-v/src/esp32c3/esp-hal-3rdparty log -1 --pretty=" * hal:%cs (%h)"
+       $(Q) imgtool dumpinfo nuttx.bin | grep img_size | awk -F ' ' '{print " * " "firmware:" 0 + $$2 " (" $$2 ")"}'
+       $(Q) objdump -h nuttx | grep .[di]ram | awk -F ' ' '{print "0x" $$3}' | awk '{s+=$$1} END {print " * " "free:" 16384 + 393216 - s "\n * " "used:" s}'
 endef
 else
 define MKIMAGE
```

Result
```
LD: nuttx                                                                       
CP: nuttx.hex
MKIMAGE: NuttX binary
imgtool sign --pad --confirm --align 4 -v 0 -s auto -H 32 --pad-header -S 0x150000 nuttx.hex nuttx.bin
Generated: nuttx.bin (MCUboot compatible)
 * nuttx:2024-10-18 (d39bcd4ad5)
 * hal:2024-06-07 (9bc2c73838)
 * firmware:1038464 (0xfd880)
 * free:296448
 * used:113152
```

metarutaiga, 2024-11-11:
I found it updated and replaced the rom codes.

| Size | Symbol | Note |
| ---------- | ---- | ---- |
|00000080|ieee80211_alloc_tx_buf||
|00000270|ieee80211_ampdu_reorder|https://github.com/espressif/esp-idf/commit/6f20668b2468baedce2bedfe24aa3687ddf04b29|
|0000019e|ieee80211_output_do||
|00000294|ieee80211_output_process|https://github.com/espressif/esp-idf/commit/039ef51db50992db68764626d5cf73a32d53d6cb|
|0000004e|ieee80211_post_hmac_tx||
|00000048|lmacAdjustTimestamp||
|000000d6|lmacDiscardFrameExchangeSequence||
|00000314|lmacEndFrameExchangeSequence|https://github.com/espressif/esp-idf/commit/edbf44b290729b5c3ce935ba1a5ba99dc4e93ee6|
|0000005c|lmacMSDUAged|https://github.com/espressif/esp-idf/commit/36e082d6e222fd29e4cdb770dd6ad37b4fffec01|
|00000138|lmacProcessTxComplete||
|00000104|lmacProcessTxSuccess||
|000001f8|lmacSetTxFrame|https://github.com/espressif/esp-idf/commit/b20922c6d582082e151f269e0ddedb5ba7a79ef8|
|00000080|ppAssembleAMPDU|https://github.com/espressif/esp-idf/commit/b7d599984bc5fc65459e8959d46efc76615ce8d4|
|000002e0|ppCalTxAMPDULength|https://github.com/espressif/esp-idf/commit/b7d599984bc5fc65459e8959d46efc76615ce8d4|
|0000006a|ppDirectRecycleAmpdu||
|00000158|ppMapTxQueue||
|0000012a|ppMapWaitTxq|https://github.com/espressif/esp-idf/commit/e6efcc407e54713bd31eae1bbdbbf3554ee6deda|
|000000b0|ppProcessLifeTime||
|00000194|ppProcTxDone||
|000000a8|ppRegressAmpdu||
|0000011c|ppRxPkt||
|00000118|wDev_AppendRxBlocks||
|000002d2|wDev_IndicateFrame||
|0000033a|wDev_ProcessRxSucData||

closed 2025-07-23 by fdcavalcanti