apache/nuttx #14281: [BUG] cmake savedefconfig failure after https://github.com/apache/nuttx/pull/13815

xuxin930, 2024-10-15:
### Description / Steps to reproduce the issue

> [!IMPORTANT]  
> cmake savedefconfig failure after https://github.com/apache/nuttx/pull/13815

## Steps to reproduce the problem
### 1. config with CMake
```shell
cmake -B build -DBOARD_CONFIG=qemu-armv7a:nsh
``` 
### 2. change any config with menuconfig
```shell
cmake --build build -t menuconfig
# For example  Disable CONFIG_EXAMPLES_HELLO
``` 
### 3. savedefconfig use cmake
```shell
cmake --build build -t savedefconfig
``` 

Issue: **Neither the BINARY directory nor the defconfig in the config directory has been modified.**
Expect: Both are modified to the latest

This issue did not exist before this PR: https://github.com/apache/nuttx/pull/13815

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 20.04

### NuttX Version

12.5.1

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Build System], [Area: Kconfig]

### Verification

- [X] I have verified before submitting the report.

xuxin930, 2024-10-15:
hi @simbit18 can you take a look at this issue?

simbit18, 2024-10-18:
hi @xuxin930 sure !!!

simbit18, 2024-10-18:
@xuxin930  Clarification

```
3. savedefconfig use cmake
cmake --build build -t savedefconfig
Issue: Neither the BINARY directory nor the defconfig in the config directory has been modified.
Expect: Both are modified to the latest

```
Why should savedefconfig replace the defconfig file in the boards/arm/qemu-armv7a/configs/nsh path?
Its task (as in compiling with make) should only be to recreate the defconfig file in the cmake buid path.
Perhaps you should have a target with a different name.

xuxin930, 2024-10-18:
yes @simbit18 
Thank you for your reminder. You are right. 
We just need to ensure that the defconfig saved in BINARY is what we expect.

simbit18, 2024-10-18:
I also created replacedefconfig https://github.com/simbit18/nuttx/commit/2061d3a4afebd4413ce6b75ae301e10d4712f20d
Do I need it or remove it?

xuxin930, 2024-10-18:
> @xuxin930 I also created replacedefconfig [simbit18@2061d3a](https://github.com/simbit18/nuttx/commit/2061d3a4afebd4413ce6b75ae301e10d4712f20d) Do I need it or remove it?

I think rename is unnecessary, as it has different semantics from Kconfig.
How about keeping the name savedefconfig?

simbit18, 2024-10-18:
@xuxin930  OK then I rename current savedefconfig to refreshsilent
and my replacedefconfig to savedefconfig confirm ?

xuxin930, 2024-10-18:
This seems to solve the problem I reported.

because after modifying the config through menuconfig on the Makefile base,
then executing savedefconfig, we can get the modified defconfig

simbit18, 2024-10-18:
OK I add PR.

raiden00pl, 2024-10-18:
> Why should savedefconfig replace the defconfig file in the boards/arm/qemu-armv7a/configs/nsh path?
Its task (as in compiling with make) should only be to recreate the defconfig file in the cmake buid path.
https://github.com/ulfalizer/Kconfiglib/blob/master/savedefconfig.py

@simbit18  look at this PR discussion: https://github.com/apache/nuttx/pull/11416
FYI @anchao

xuxin930, 2024-10-18:
> @simbit18 look at this PR discussion: #11416 FYI @anchao

thanks @raiden00pl 
our original intention is to be consistent with the behavior of Makefile. 
That is, only save the defconfig of BINARY_DIR, not the original defconfig of the board.

closed 2024-10-18 by xiaoxiang781216