apache/nuttx #14374: [BUG] ci output "/usr/bin/bash: line 1: p32-g++: command not found"

xiaoxiang781216, 2024-10-17:
### Description / Steps to reproduce the issue

here is one of output:
https://github.com/apache/nuttx/actions/runs/11365260351/job/31613141510?pr=14364

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu

### NuttX Version

master

### Issue Architecture

[Arch: avr]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-17:
@lupyuen could you look at this problem?

lupyuen, 2024-10-18:
Seems to be caused by [tools/refresh.sh](https://github.com/apache/nuttx/blob/master/tools/refresh.sh). The same kind of messages appear when we run `tools/refresh.sh` in our CI. Will track down some more.

```text
./tools/refresh.sh --silent c5471evm/httpd
  Normalize c5471evm/httpd
/usr/bin/bash: line 1: arm-nuttx-eabi-g++: command not found
```

https://github.com/lupyuen5/label-nuttx/actions/runs/11343543363/job/31546412794

lupyuen, 2024-10-28:
Sorry @simbit18: Would you have any idea why [tools/refresh.sh](https://github.com/apache/nuttx/blob/master/tools/refresh.sh) is showing this message? Thanks!

simbit18, 2024-10-28:
@lupyuen if I'm not mistaken it seems to be solved
here is not present
https://github.com/apache/nuttx/actions/runs/11549938848/job/32143990355?pr=14528

```
====================================================================================
Configuration/Tool: flipnclick-pic32mz/nsh,CONFIG_MIPS32_TOOLCHAIN_PINGUINOL
2024-10-28 08:33:57
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Disabling CONFIG_MIPS32_TOOLCHAIN_GNU_ELF
  Enabling CONFIG_MIPS32_TOOLCHAIN_PINGUINOL
  Building NuttX...
  Normalize flipnclick-pic32mz/nsh
====================================================================================
Configuration/Tool: flipnclick-pic32mz/nxlines,CONFIG_MIPS32_TOOLCHAIN_PINGUINOL
2024-10-28 08:34:33
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Disabling CONFIG_MIPS32_TOOLCHAIN_GNU_ELF
  Enabling CONFIG_MIPS32_TOOLCHAIN_PINGUINOL
  Building NuttX...
  Normalize flipnclick-pic32mz/nxlines
====================================================================================

```

lupyuen, 2024-10-28:
Thanks @simbit18! Wonder if this message could be caused by the same issue?
```text
Configuration/Tool: c5471evm/nsh,CONFIG_ARM_TOOLCHAIN_GNU_EABI
/usr/bin/bash: line 1: arm-nuttx-eabi-gcc: command not found
```
https://github.com/NuttX/nuttx/actions/runs/11535897153/job/32116015943#step:7:242

xiaoxiang781216, 2024-10-29:
should we hide this harmless warning from script.

simbit18, 2024-10-29:
@xiaoxiang781216 I don't think so.

xiaoxiang781216, 2024-10-29:
why not? or we add nuttx toolchain to docker image?

simbit18, 2024-10-29:
> or we add nuttx toolchain to docker image?

This might be a good alternative to remove the warning. However, being busy with my work, I have little time to test and add in the docker image.