apache/nuttx #14376: [URGENT] Reducing our usage of GitHub Runners

lupyuen, 2024-10-17:
### Is your feature request related to a problem? Please describe.

TODO

### Describe the solution you'd like

TODO

PRs that affect all architectures (Arm, RISC-V, Xtensa, etc) shall run only half the jobs (now we will run only arm-01 to arm-07, previously arm-01 to arm-14)

When merged, we will still run all jobs arm-01 to arm-14

4 Scheduled Merge Jobs per day

Windows and macOS

Analysis of jobs in past 24 hours: https://docs.google.com/spreadsheets/d/1ujGKmUyy-cGY-l1pDBfle_Y6LKMsNp7o3rbfT1UkiZE/edit?gid=0#gid=0

### Verification

- [X] I have verified before submitting the report.

raiden00pl, 2024-10-17:
One of the methods proposed by, if I remember correctly @btashton, is to replace many simple configurations  for some boards (mostly for peripherals testing) with one large `jumbo` config activating everything possible.
This won't work for chips with low memory, but it will save some CI resources anyway.

lupyuen, 2024-10-17:
@raiden00pl Yep I agree. Or we could test a complex target like `board:lvgl`?

lupyuen, 2024-10-17:
Here's another comment about macOS and Windows: https://github.com/apache/nuttx/pull/14377#issuecomment-2418914068

yamt, 2024-10-17:
sorry, let me ask a dumb question.
what plan are we using? https://github.com/pricing
is apache paying for it?

yamt, 2024-10-17:
> > what plan are we using? https://github.com/pricing
> 
> @yamt It's probably a special plan negotiated by ASF and GitHub? It's not mentioned in the ASF Policy for GitHub Actions: https://infra.apache.org/github-actions-policy.html

do you know if the macos/windows premium applies as usual?
the policy page seems to have no mention about it.

> I find this "contract" a little strange. Why are all ASF Projects subjected to the same quotas? And why can't we increase the quota if we happen to have additional funding?

yea, i guess projects have very different sizes/demands.
(i feel nuttx is using too much anyway though :-)

TimJTi, 2024-10-17:
> ...I'd rather use my own Mac Mini to execute the NuttX CI Tests, once a day?
Is there any merit in "farming out" CI tests to those with boards? I think there was a discussion about NuttX owning a suite of boards but not sure where that got to - and would depend on just 1 or 2 people managing it. 

As an aside, is there a guide to self-running CI? As I work on a custom board it would be good for me to do this occasionally but I have noi idea where to start!

lupyuen, 2024-10-17:
@TimJTi Here's how I do daily testing on Milk-V Duo S SBC: https://lupyuen.github.io/articles/sg2000a

TimJTi, 2024-10-17:
> @TimJTi Here's how I do daily testing on Milk-V Duo S SBC: https://lupyuen.github.io/articles/sg2000a

And I just RTFM...the "official" guide is [here](https://nuttx.apache.org/docs/latest/guides/citests.html) so I'll review both and hopefully get it working - and submit any tweaks/corrections/enhancements I find are needed to the NuttX "How To" documentation

jerpelea, 2024-10-17:
[like]  Jerpelea, Alin reacted to your message:

michallenc, 2024-10-17:
> And I just RTFM...the "official" guide is [here](https://nuttx.apache.org/docs/latest/guides/citests.html) so I'll review both and hopefully get it working - and submit any [...]

These works, but it does not describe the entire CI, just how to run pytest checks for `sim:citest` configuration.

cederom, 2024-10-17:
Yes let's cut what we can (but to keep at least minimal functional configure, build, syntax testing) and see what are the cost reduction. We need to show Apache we are working on the problem. So far optimitzations did not cut the use and we are in danger of loosing all CI :-(

On the other hand that seems not fair to share the same CI quota as small projects. NuttX is a fully featured RTOS working on ~1000 different devices. In order to keep project code quality we need the CI.

Maybe its time to rethink / redesign from scratch the CI test architecture and implementation?

cederom, 2024-10-17:
Another problem is that people very often send unfinished undescribed PRs that are updated without a comment or request that triggers whole big CI process several times :-( 

Some changes are sometimes required sometimes and we cannot avoid that. But maybe we can make something more "adaptive" so only minimal CI is launched by default, preferably only in area that was changed, then with all approvals we can make one final big check before merge?

lupyuen, 2024-10-17:
@cederom Maybe our PRs should have a Mandatory Field: Which NuttX Config to build, e.g. `rv-virt:nsh`. Then the CI Workflow should do `tools/configure.sh rv-virt:nsh && make`. Before starting the whole CI Build?

cederom, 2024-10-17:
> @cederom Maybe our PRs should have a Mandatory Field: Which NuttX Config to build, e.g. `rv-virt:nsh`. Then the CI Workflow should do `tools/configure.sh rv-virt:nsh && make`. Before starting the whole CI Build?

People often cant fill even one single sentence to describe Summary, Impact, Testing :D This may be detected automatically.. or we can just see what architecture is the cheapest one and use it for all basic tests..?

raiden00pl, 2024-10-17:
> Another problem is that people very often send unfinished undescribed PRs that are updated without a comment or request that triggers whole big CI process several times :-(

Often contributors use CI to test all configuration instead of testing changes locally. On one hand I understand this because compiling all configurations on a local machine takes a lot of time, on the other hand I'm not sure if CI is for this purpose (especially when we have limits on its use).

> @cederom Maybe our PRs should have a Mandatory Field: Which NuttX Config to build, e.g. rv-virt:nsh. Then the CI Workflow should do tools/configure.sh rv-virt:nsh && make. Before starting the whole CI Build?

It won't work. Users are lazy, and in order to choose what needs to be compiled correctly, you need a comprehensive knowledge of the entire NuttX, which is not that easy. 
The only reasonable option is to automate this process.

cederom, 2024-10-17:
So it looks like for now, where dramatic steps need to be taken, we need to mark all PR as drafts and start CI by hand when we are sure all is ready for merge? o_O

lupyuen, 2024-10-17:
Stats for the past 24 hours: We consumed __61 Full-Time Runners__, still got a long way away from our target of 25 Full-Time Runners (otherwise ASF will halt our servers in 12 days)
- We have switched to [Four Scheduled Merge Jobs](https://gist.github.com/lupyuen/9c3ecea33525f66e163d31ddda85dba2) per day
- `nuttx-apps` has stopped macOS and Windows Jobs. But not much impact, since we don't compile `nuttx-apps` often <br> https://github.com/apache/nuttx-apps/pull/2750
- Still waiting for `nuttx` repo to stop macOS and Windows Jobs, we're stuck at a disagreement <br> https://github.com/apache/nuttx/pull/14377
- Also waiting for `nuttx` repo to Halve The Jobs <br> https://github.com/apache/nuttx/pull/14386
- And for `nuttx-apps` to Halve The Jobs (probably not much impact, since we don't compile `nuttx-apps` often) <br> https://github.com/apache/nuttx-apps/pull/2753
- Will wait for the above to be merged, then we monitor some more

[image]

cederom, 2024-10-18:
Okay its 0000UTC. We are really short on time. I have merged changes. Lets monitor the use now for 24h we need metrics. We can always revert the commits.

cederom, 2024-10-18:
Sorry no clue why it closed in my name o_O

lupyuen, 2024-10-18:
The builds are so much faster today yay! https://github.com/apache/nuttx/actions/runs/11395811301
[image]

michallenc, 2024-10-18:
We can also disable CI checks for draft merge requests. I think it doesn't make much sense to run them as further commits/force pushes are expected.

lupyuen, 2024-10-18:
Something That Bugs Me: __Timeout Errors__ will cost us precious GitHub Minutes. The remaining jobs get killed, and restarting these remaining jobs from scratch will consume extra GitHub Minutes.
(1) How do we retry these Timeout Errors?
(2) Can we have Restartable Builds? Doesn't quite make sense to build everything from scratch (arm6, arm7, riscv7) just because one job failed (xtensa2)
(3) Or xtensa2 should wait for others to finish, before it declares a timeout and dies? Hmmm...

```text
Configuration/Tool: esp32s2-kaluga-1/lvgl_st7789
curl: (28) Failed to connect to github.com port 443 after 133994 ms: Connection timed out
```
https://github.com/apache/nuttx/actions/runs/11395811301/attempts/1

cederom, 2024-10-18:
> @lupyuen: (2) Can we have Restartable Builds? Doesn't quite make sense to build everything from scratch (arm6, arm7, riscv7) just because one job failed (xtensa2)

It is possible to restart only failed tasks on GitHub :-)

If you mean the task could restart where it left.. I am not sure its possible because underlying configuration could change after update to be verified and things need to be started from scratch? :-)

lupyuen, 2024-10-18:
__11 Days To Doomsday:__ But we're doing much better already! In the past 24 hours, we consumed __36 Full-Time GitHub Runners__. We're getting closer to the ASF Target of __25 Full-Time Runners__! Today we shall:

- Halve the Jobs for __RISC-V, Xtensa and Simulator__ for Complex PRs <br>
  https://github.com/apache/nuttx/pull/14400

- Do the same for `nuttx-apps` repo <br>
  https://github.com/apache/nuttx-apps/pull/2758

- Reduce the Scheduled Merge Jobs to __Two Per Day__ at 00:00 / 12:00 UTC (down from Four Per Day)

Hopefully we'll reach the ASF Target tomorrow, and ASF won't kill our servers no more! Thanks!

[image]

lupyuen, 2024-10-19:
_When NuttX merges our PR, the Merge Job won't run until 00:00 UTC and 12:00 UTC. How can we be really sure that our PR was merged correctly?_

Let's create a __GitHub Org__, fork the NuttX Repo and trigger the __CI Workflow__. (Which won't charge any extra GitHub Runner Minutes to NuttX Project!)

- https://github.com/apache/nuttx/issues/14407

(I think this might also work if ASF shuts down our CI Servers. We can create many many orgs actually)

cederom, 2024-10-19:
Sounds like a repo clone that will verify nuttx and nuttx-apps master independently twice a day?

lupyuen, 2024-10-19:
@cederom You read my mind :-)

lupyuen, 2024-10-19:
Hi All: Our [__Merge Jobs are now at github.com/nuttxpr/nuttx__](https://github.com/nuttxpr/nuttx/actions)

Yesterday we spent __One-Third__ of our GitHub Runner Minutes on Merge Jobs. This is not sustainable, so I moved them to `nuttxpr` repo. (Which won't be charged)

[image]

[The data from yesterday](https://docs.google.com/spreadsheets/d/1ujGKmUyy-cGY-l1pDBfle_Y6LKMsNp7o3rbfT1UkiZE/edit?gid=650325940#gid=650325940) shows that our Scheduled Merge Job keeps getting disrupted by other Merged PRs. And when we restart a Scheduled Merge Job, we waste GitHub Minutes. (101 GitHub Hours for the Scheduled Merge Job!)

__Two-Thirds__ of our GitHub Runner Minutes were spent on Creating and Updating PRs. That's why we're skipping half the jobs today.

Hi @xiaoxiang781216 @GUIDINGLI @cederom: With immediate effect, please see [__github.com/nuttxpr/nuttx__](https://github.com/nuttxpr/nuttx/actions) for our Merge Jobs. I will trigger the jobs daily at 00:00 UTC and 12:00 UTC. I have given you Admin Access to `nuttxpr` in case you need to restart the jobs. Thanks!

cederom, 2024-10-19:
Thank you @lupyuen !! :-) We have the https://github.com/nuttx organization too maybe we can make use of it too? :-)

lupyuen, 2024-10-19:
__10 Days to Shutdown:__ Or maybe not, because our GitHub Usage has dropped to __5 Full-Time Runners yay!__ Let's keep this below __25 Full-Time Runners,__ and make ASF super happy!

- Today we split the Build Jobs for __Arm64 and x86_64__ from `other`, making them faster (Hello Goldfish :-)

  https://github.com/apache/nuttx/pull/14282

- Our [__Scheduled Merge Jobs are now at github.com/nuttxpr/nuttx__](https://github.com/nuttxpr/nuttx/actions), which I trigger manually at 00:00 UTC and 12:00 UTC daily. I'm still [running this script](https://github.com/lupyuen/nuttx-release/blob/main/kill-push-master.sh), forever killing any Merge Jobs at `nuttx` and `nuttx-apps` repos.

- Excellent Initiative by @raiden00pl: We __Merge Multiple Targets__ into One Target, and reduce the build time

  https://github.com/apache/nuttx/pull/14410

- Here's the [Email I sent to ASF Infra Team](https://gist.github.com/lupyuen/7442be51444a57e0d9c32986b4500f8a)

- Then Again: Yesterday was quiet, with few PRs. (But it's great for analysing the data!)

Thank you so much for your patience, let's keep this up! 🙏

[image]

lupyuen, 2024-10-20:
> We have the https://github.com/nuttx organization too maybe we can make use of it too? :-)

I think we learnt a Painful Lesson today: Freebies Won't Last Forever! The new GitHub Org for NuttX should probably be a __Paid GitHub Org__:
- New GitHub Org shall be sponsored by our generous Stakeholder Companies (Espressif, Sony, Xiaomi, ...)
- New GitHub Org shall be maintained by a Paid Employee of our Stakeholder Companies
- Which means clicking Twice Per Day to trigger the Scheduled Merge Jobs (I'm getting tired of this)
- And restarting the Scheduled Merge Job (if it fails). Also: [Killing the Old Merge Jobs](https://github.com/lupyuen/nuttx-release/blob/main/kill-push-master.sh)
- New GitHub Org shall host the Official Downloads of NuttX Compiled Binaries (for our upcoming Board Testing Farm)
- New GitHub Org will eventually offload more CI Jobs from our GitHub Repos (e.g. macOS and Windows Builds)

xiaoxiang781216, 2024-10-20:
@lupyuen could we select arm-01, arm-03, arm-05... instead of arm-01, arrm-02,... arm-07, which could improve the chip coverage?

lupyuen, 2024-10-20:
> ould we select arm-01, arm-03, arm-05... instead of arm-01, arrm-02,... arm-07, which could improve the chip coverage?

@xiaoxiang781216 Good idea. RP2040 is in [arm-06](https://github.com/apache/nuttx/blob/master/tools/ci/testlist/arm-06.dat), should we select [arm-06](https://github.com/apache/nuttx/blob/master/tools/ci/testlist/arm-06.dat) instead of
[arm-13](https://github.com/apache/nuttx/blob/master/tools/ci/testlist/arm-13.dat)?

So we build arm-01, arm-03, arm-05, arm-06, arm-07, arm-09, arm-11?

cederom, 2024-10-20:
I am not sure if we should diverge from Apache that much, but its also sad to see zero Apache support when/where we need them most.

Having financial support from companies that use NuttX would be nice, but should we use it to pay GitHub or better pay freelancers to develop more NuttX code? GitHub benefits all here, while NuttX should.

Maybe we should look for alternatives to GitHub like GitLab where other OS runners are also possible as a complementary / backup solution to GitHub?

And the independent decentralized testing farms seems the best alternative. But its a long road.

I just hope all this won't impact NuttX quality.

lupyuen, 2024-10-20:
Yeah it doesn't sound right that an Unpaid Volunteer is monitoring our CI Servers 24 x 7 🤔
[image]

xiaoxiang781216, 2024-10-20:
> @xiaoxiang781216 Good idea. RP2040 is in [arm-06](https://github.com/apache/nuttx/blob/master/tools/ci/testlist/arm-06.dat), should we select [...]

sure.

lupyuen, 2024-10-20:
__9 Days to Nirvana:__ Yesterday (quiet Sunday) we consumed __3 Full-Time GitHub Runners__. Let's keep this below __25 Full-Time Runners,__ and we'll break free from suffering!

- No major changes for today, we'll watch and monitor

- Scheduled Merge Jobs are still at [nuttxpr/nuttx](https://github.com/nuttxpr/nuttx/actions/workflows/build.yml)

- Once in a while: I'll run Ad Hoc Merge Jobs for NuttX Apps at [nuttxpr/nuttx-apps](https://github.com/nuttxpr/nuttx-apps/actions/workflows/build.yml) (Kinda redundant because `nuttx` will also compile `nuttx-apps`)

- Today is Monday, we expect the load to increase. Hoping for the best, thanks everyone!
 
[image]

lupyuen, 2024-10-21:
_What if we could run the CI Jobs on our own Ubuntu PCs? Without any help from GitHub Actions?_

I'm experimenting with a "Build Farm" at home (refurbished PC) that __runs NuttX CI Jobs all day__ non-stop 24 x 7:
- Check out `master` branch of `nuttx`, run CI Job `arm-01`
- Wait for `arm-01` to complete (roughly 1.5 hours)
- Check out `master` branch of `nuttx`, run CI Job `arm-02`
- Wait for `arm-02` to complete (roughly 1.5 hours)
- Do the same until `arm-14`, then loop back to `arm-01`
- [Here's the CI Output Log](https://gist.github.com/nuttxpr)

How does it work?
- [run-job.sh](https://github.com/lupyuen/nuttx-release/blob/main/run-job.sh) will run a single CI Job, by calling the NuttX Docker Image, which is called by...
- [run-ci.sh](https://github.com/lupyuen/nuttx-release/blob/main/run-ci.sh) looping forever through `arm-01` to `arm-14`, running the job, searching for errors and uploading the logs

[image]

lupyuen, 2024-10-21:
__8 Days to Diwali:__ Will our CI Servers go Dark? Sorry we're not sure, because the ASF Infra Reports are Down (sigh). But I think we briefly hit a peak of __21 Full-Time GitHub Runners__, which is still within our target of __25 Full-Time Runners__.

- Simulator CI Job `sim-01` has become the bottleneck. Today We'll split `sim-01` and add `sim-03`:

  https://github.com/apache/nuttx/pull/14428

  https://github.com/apache/nuttx-apps/pull/2763

- Since The ASF Reports Are Down: I'll try calling the [GitHub API](https://docs.github.com/en/rest/actions/workflow-runs?apiVersion=2022-11-28#get-workflow-run-usage) to fetch the Elapsed Duration of every job. Then extrapolate to GitHub Runner Minutes.

[image]

lupyuen, 2024-10-22:
ASF Infra Reports are still now. But now we have our own __Live Metrics for Full-Time GitHub Runners!__ (reload for updates)

[image]

This shows the number of __Full-Time Runners for the Day__, computed since 00:00 UTC. (Remember: We should keep this below 25) How it works:
- [compute-github-runners.sh](https://github.com/lupyuen/nuttx-release/blob/main/compute-github-runners.sh) calls GitHub API to add up the Duration of All GitHub Jobs for today. Then it extrapolates the Number of Full-Time GitHub Runners. (1 Elapsed Hour roughly equals 8 GitHub Runner Hours)
- [run.sh](https://github.com/lupyuen/nuttx-metrics/blob/main/run.sh) calls the script above to render the Full-Time GitHub Runners as a PNG (with ImageMagick)

xiaoxiang781216, 2024-10-22:
@lupyuen the new number is very small, should we try restoring some macOS/msys2/windows ci?

lupyuen, 2024-10-22:
@xiaoxiang781216 Let's monitor for the rest of the day. Towards the end of the day, the number of Full-Time Runners will probably jump to 21. (Like yesterday)

We could mirror the NuttX Repo to another GitHub Org account and run the Windows and macOS Jobs there (so they won't add to our quota). We just need minor changes to `build.yml` and `arch.yml`: https://github.com/apache/nuttx/issues/14407 (at the bottom of the doc)

xiaoxiang781216, 2024-10-22:
could we use this https://github.com/nuttx account?

lupyuen, 2024-10-22:
@xiaoxiang781216 Yep lemme work on it

lupyuen, 2024-10-22:
@xiaoxiang781216 The macOS and Windows Builds are now running in our NuttX Mirror: https://github.com/NuttX/nuttx/actions/runs/11452528962

I made 2 fixes to enable the macOS and Windows Builds: [build.yml](https://github.com/NuttX/nuttx/commit/51b943cf78c5bc8fc6b876e524cf00f17790554d) and [arch.yml](https://github.com/NuttX/nuttx/commit/98ac7882d2b583abedb487cc601f6eb47f86a7b0)

Lemme figure out how to automate this 🤔

simbit18, 2024-10-22:
@lupyuen Thank you for this amazing work!

lupyuen, 2024-10-22:
__7 Days to Event Horizon:__ Yesterday we consumed __4 Full-Time GitHub Runners__ (too low?). Let's keep this number below __25 Full-Time Runners__, so our CI Servers won't sink into the black hole!

- Last 5 days: We consumed __13 Full-Time Runners__. That's below ASF's Limit of 25 Full Time Runners. So we're still on target! I shared the good news with ASF Infra Team [in my email](https://gist.github.com/lupyuen/45095ade8531340dd6932e24a4119800)

- Which means I shall __Freeze our CI Workflow__. Wait for ASF Infra Team to approve on 30 Oct. Then we decide how to reorg our Build Targets.

- Our [__Merge Jobs are now at github.com/NuttX/nuttx__](https://github.com/NuttX/nuttx/actions/workflows/build.yml). (Includes macOS and Windows Builds)

- How It Happens: Daily at 00:00 UTC and 12:00 UTC, I click `Sync Fork > Discard Commits`. Then I [run this script](https://github.com/lupyuen/nuttx-release/blob/main/enable-macos-windows.sh) to enable the macOS and Windows Builds.

- Don't Forget: I'm still [running this script](https://github.com/lupyuen/nuttx-release/blob/main/kill-push-master.sh) to kill the Merge Jobs on the old `nuttx` and `nuttx-apps` repos. (Otherwise the GitHub Runners will spike!)

The numbers from yesterday are a little sus, lemme verify...

[image]

Alternatively: We have our cool purple report (14 Full-Time Runners for yesterday)...

[image]

[(Live Image)](https://lupyuen.github.io/nuttx-metrics/github-fulltime-runners.png) [(Live Log)](https://github.com/lupyuen/nuttx-metrics/blob/main/compute-github-runners.log)

Thanks everyone for the kudo! :-)

lupyuen, 2024-10-23:
__6 Days to Serenity:__ Yesterday we consumed __13 Full-Time GitHub Runners__. (That's half of the ASF Quota for GitHub Runners) Everything is coming up nicely (even ASF agrees), thanks to everyone! 🙏

[image]

Our Cool Purple Report now turns Orange when the load goes up (yep our estimated data is a little higher than the ASF data, just to be safe)...

[image]

[(Live Image)](https://lupyuen.github.io/nuttx-metrics/github-fulltime-runners.png) [(Live Log)](https://github.com/lupyuen/nuttx-metrics/blob/main/compute-github-runners.log)

lupyuen, 2024-10-24:
__5 Days to Freedom:__ Yesterday we consumed __14 Full-Time GitHub Runners__. That's 56% of the ASF Quota for Full-Time Runners. Looking good!

- We peaked briefly at 26 Full-Time Runners. Needs more monitoring.

- Awesome Workflow by @simbit18: Run the Windows Build on any NuttX Repo...

  https://github.com/apache/nuttx/pull/14499#issuecomment-2435359296

- Data from yesterday:

  [image]

- Let's keep our report Cool and Purple for today. Thank you so much!

  [image]

  [(Live Image)](https://lupyuen.github.io/nuttx-metrics/github-fulltime-runners.png) [(Live Log)](https://github.com/lupyuen/nuttx-metrics/blob/main/compute-github-runners.log)

stbenn, 2024-10-25:
@lupyuen It looks I made a mistake with some commit messages, that caused our branch to get referenced to a few issues in the apache repo. My apologies. I believe I have removed the commit message references, but if there is anything else I need to do to fix this, please let me know and I will get on it ASAP.

lupyuen, 2024-10-25:
__4 Days to Festivity:__ Yesterday we consumed __13 Full-Time GitHub Runners__ (half of the ASF Quota for GitHub Runners)...

[image]

__Past 7 Days:__ We also used an average of __13 Full-Time GitHub Runners__...

[image]

So we're on track to make ASF very happy on 30 Oct! Let's monitor today...

[image]

[(Live Image)](https://lupyuen.github.io/nuttx-metrics/github-fulltime-runners.png) [(Live Log)](https://github.com/lupyuen/nuttx-metrics/blob/main/compute-github-runners.log)

cederom, 2024-10-26:
Thank you @lupyuen for your amazing work!! Have a good calm weekend :-) :-)

lupyuen, 2024-10-26:
__3 Days to Tranquility:__ Yesterday was a quiet Saturday (no more Release Builds yay!). We consumed only __4 Full-Time GitHub Runners__...

[image]

Let's hope today will be a peaceful Sunday...

[image]

[(Live Image)](https://lupyuen.github.io/nuttx-metrics/github-fulltime-runners.png) [(Live Log)](https://github.com/lupyuen/nuttx-metrics/blob/main/compute-github-runners.log)

lupyuen, 2024-10-27:
Something strange about __Network Timeouts__ in our Docker Workflows: First Run fails while [downloading something from GitHub](https://github.com/nuttxpr/nuttx/actions/runs/11535899222/job/32111488205#step:7:626):
```text
Configuration/Tool: imxrt1050-evk/libcxxtest,CONFIG_ARM_TOOLCHAIN_GNU_EABI
curl: (28) Failed to connect to github.com port 443 after 134188 ms: Connection timed out
make[1]: *** [libcxx.defs:28: libcxx-17.0.6.src.tar.xz] Error 28
```

Second Run fails again, while [downloading NimBLE from GitHub](https://github.com/nuttxpr/nuttx/actions/runs/11535899222/job/32112716849#step:7:536):
```text
Configuration/Tool: nucleo-wb55rg/nimble,CONFIG_ARM_TOOLCHAIN_GNU_EABI
curl: (28) Failed to connect to github.com port [443](https://github.com/nuttxpr/nuttx/actions/runs/11535899222/job/32112716849#step:7:444) after 134619 ms: Connection timed out
make[2]: *** [Makefile:55: /github/workspace/sources/apps/wireless/bluetooth/nimble_context] Error 2
```

[Third Run succeeds.](https://github.com/nuttxpr/nuttx/actions/runs/11535899222) Why do we keep seeing these errors: GitHub Actions with Docker, can't connect to GitHub itself?

Is something misconfigured in our Docker Image? But the exact same Docker Image runs fine on [my own Build Farm](https://lupyuen.codeberg.page/articles/ci2.html). It [doesn't show any errors](https://lupyuen.codeberg.page/articles/ci2.html).

Is GitHub Actions starting our Docker Container with the wrong MTU (Network Packet Size)? 🤔
- https://github.com/actions/actions-runner-controller/issues/393
- [Docker MTU issues and solutions](https://mlohr.com/docker-mtu/)

lupyuen, 2024-10-27:
__2 Days to Transcendence:__ Yesterday we consumed __10 Full-Time GitHub Runners__. We peaked briefly at 21 while compiling a few NuttX Apps.

[image]

Let's keep on monitoring thanks!

[image]

[(Live Image)](https://lupyuen.github.io/nuttx-metrics/github-fulltime-runners.png) [(Live Log)](https://github.com/lupyuen/nuttx-metrics/blob/main/compute-github-runners.log)

lupyuen, 2024-10-28:
### Monitoring our CI Servers 24 x 7

This runs on my 4K TV (Xiaomi 65-inch) all day, all night:

[image]

When I'm out on [Overnight Hikes](https://www.strava.com/activities/12737067287): I check my phone at every water break:
[image]

I have GitHub Scripts that will run on Termux Android:
- [enable-macos-windows2.sh](https://github.com/lupyuen/nuttx-release/blob/main/enable-macos-windows2.sh): Enable the macOS and Windows Builds
- [compute-github-runners2.sh](https://github.com/lupyuen/nuttx-release/blob/main/compute-github-runners2.sh): Compute the number of Full-Time GitHub Runners for the day

cederom, 2024-10-28:
Lup's Operations Center =)

lupyuen, 2024-10-28:
__1 Day to Utopia:__ Yesterday was a busy Monday, we consumed __14 Full-Time GitHub Runners__. That's 56% of the ASF Quota for Full-Time Runners...

[image]

We peaked briefly at 26 Full-Time Runners. Let's hang in there thanks! :-)

[image]

[(Live Image)](https://lupyuen.github.io/nuttx-metrics/github-fulltime-runners.png) [(Live Log)](https://github.com/lupyuen/nuttx-metrics/blob/main/compute-github-runners.log)

cederom, 2024-10-28:
2 days but we should be fine thanks to our Super Hero @lupyuen !! AVE =)

lupyuen, 2024-10-28:
Thank you so much @cederom! :-)

[7 more replies]

closed 2024-10-31 by lupyuen