apache/nuttx #14390: [BUG] `cp -T` is used by kconfig-merge / tools/esp32s3/Config.mk

yamt, 2024-10-18:
### Description / Steps to reproduce the issue

try building esp32s3-devkit:knsh on macOS.
it uses kconfig-merge, which internally uses `cp -T`, which is not available for non-gnu cp.
https://bitbucket.org/nuttx/tools/src/9484147c12d051014f854852d59c21d75a9616bd/kconfig-frontends/utils/kconfig-merge#lines-135

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

14.7

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Kconfig]

### Verification

- [X] I have verified before submitting the report.

fdcavalcanti, 2024-10-21:
Hi @yamt. This is a kconfig-merge issue and not exactly a ESP32S3 one, but I get the problem.
Installing `coreutils` may solve your problem. You would have access to the expected `cp` options.
Related discussion: https://gist.github.com/skyzyx/3438280b18e4f7c490db8a2a2ca0b9da

yamt, 2024-10-21:
> Hi @yamt. This is a kconfig-merge issue and not exactly a ESP32S3 one, but I get the problem.

iirc, this kconfig-merge fork is maintained by nuttx. thus i filed this issue here.

> Installing `coreutils` may solve your problem. You would have access to the expected `cp` options. Related discussion: https://gist.github.com/skyzyx/3438280b18e4f7c490db8a2a2ca0b9da

IMO, nuttx should not require coreutils.

> Did you follow those steps on NuttX [installing guide](https://nuttx.apache.org/docs/latest/quickstart/install.html#kconfig-frontend) for MacOS?

no.
(well, i might have followed an old version of these instructions. i don't remember.)

tmedicci, 2024-10-22:
This is not an issue of ESP32-S3, but for NuttX. Did you test with `kconfiglib` [too](https://nuttx.apache.org/docs/latest/quickstart/install.html#kconfig-frontend)?

How did the CI test on MacOS?