apache/nuttx #14394: help: help title

FlyyingPiggy2020, 2024-10-18:
### Description

BL602 can't be compiled accroding to the instructions on this page [here](https://nuttx.apache.org/docs/latest/platforms/risc-v/bl602/index.html),I tried changing the toolchain to version 10.2.0 ,but it still doesn't compiled.
```
lxf@lxf-Inspiron-15-5510:~/nuttxspace/nuttx$ riscv64-unknown-elf-g
riscv64-unknown-elf-g++               riscv64-unknown-elf-gcc-ranlib        riscv64-unknown-elf-gdb-add-index
riscv64-unknown-elf-gcc               riscv64-unknown-elf-gcov              riscv64-unknown-elf-gdb-add-index-py
riscv64-unknown-elf-gcc-10.2.0        riscv64-unknown-elf-gcov-dump         riscv64-unknown-elf-gdb-py
riscv64-unknown-elf-gcc-ar            riscv64-unknown-elf-gcov-tool         riscv64-unknown-elf-gprof
riscv64-unknown-elf-gcc-nm            riscv64-unknown-elf-gdb               
lxf@lxf-Inspiron-15-5510:~/nuttxspace/nuttx$ riscv64-unknown-elf-g++ -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-g++
COLLECT_LTO_WRAPPER=/home/lxf/bl_iot_sdk/toolchain/riscv/Linux/bin/../libexec/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/build/riscv64-unknown-elf-gcc/riscv-gcc/configure --target=riscv64-unknown-elf --host=x86_64-linux-gnu --prefix=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-10.2.0-2020.12.8-x86_64-linux-ubuntu14 --with-pkgversion='SiFive GCC-Metal 10.2.0-2020.12.8' --with-bugurl=https://github.com/sifive/freedom-tools/issues --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-sysroot=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-10.2.0-2020.12.8-x86_64-linux-ubuntu14/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../riscv-gcc --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc CFLAGS=-O2 CXXFLAGS=-O2 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (SiFive GCC-Metal 10.2.0-2020.12.8) 
lxf@lxf-Inspiron-15-5510:~/nuttxspace/nuttx$ make
make                        make-first-existing-target  make-ssl-cert               
makeconv                    makeinfo                    
lxf@lxf-Inspiron-15-5510:~/nuttxspace/nuttx$ make distclean 
lxf@lxf-Inspiron-15-5510:~/nuttxspace/nuttx$ tools/configure.sh bl602evb:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/lxf/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/lxf/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/lxf/nuttxspace/apps/platform/dummy
LN: include/arch to arch/risc-v/include
LN: include/arch/board to /home/lxf/nuttxspace/nuttx/boards/risc-v/bl602/bl602evb/include
LN: drivers/platform to /home/lxf/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/lxf/nuttxspace/nuttx/arch/risc-v/include/bl602
LN: arch/risc-v/src/chip to /home/lxf/nuttxspace/nuttx/arch/risc-v/src/bl602
LN: arch/risc-v/src/board to /home/lxf/nuttxspace/nuttx/boards/risc-v/bl602/bl602evb/src
mkkconfig in /home/lxf/nuttxspace/apps/audioutils
mkkconfig in /home/lxf/nuttxspace/apps/benchmarks
mkkconfig in /home/lxf/nuttxspace/apps/boot
mkkconfig in /home/lxf/nuttxspace/apps/canutils
mkkconfig in /home/lxf/nuttxspace/apps/crypto
mkkconfig in /home/lxf/nuttxspace/apps/database
mkkconfig in /home/lxf/nuttxspace/apps/examples/mcuboot
mkkconfig in /home/lxf/nuttxspace/apps/examples/module
mkkconfig in /home/lxf/nuttxspace/apps/examples/sotest
mkkconfig in /home/lxf/nuttxspace/apps/examples
mkkconfig in /home/lxf/nuttxspace/apps/fsutils
mkkconfig in /home/lxf/nuttxspace/apps/games
mkkconfig in /home/lxf/nuttxspace/apps/graphics
mkkconfig in /home/lxf/nuttxspace/apps/industry
mkkconfig in /home/lxf/nuttxspace/apps/inertial
mkkconfig in /home/lxf/nuttxspace/apps/interpreters/luamodules
mkkconfig in /home/lxf/nuttxspace/apps/interpreters
mkkconfig in /home/lxf/nuttxspace/apps/logging
mkkconfig in /home/lxf/nuttxspace/apps/lte
mkkconfig in /home/lxf/nuttxspace/apps/math
mkkconfig in /home/lxf/nuttxspace/apps/mlearning
mkkconfig in /home/lxf/nuttxspace/apps/netutils
mkkconfig in /home/lxf/nuttxspace/apps/sdr
mkkconfig in /home/lxf/nuttxspace/apps/system
mkkconfig in /home/lxf/nuttxspace/apps/testing
mkkconfig in /home/lxf/nuttxspace/apps/videoutils
mkkconfig in /home/lxf/nuttxspace/apps/wireless/bluetooth
mkkconfig in /home/lxf/nuttxspace/apps/wireless/ieee802154
mkkconfig in /home/lxf/nuttxspace/apps/wireless
mkkconfig in /home/lxf/nuttxspace/apps
Loaded configuration '.config'
Configuration saved to '.config'
lxf@lxf-Inspiron-15-5510:~/nuttxspace/nuttx$ make
Create version.h
LN: platform/board to /home/lxf/nuttxspace/apps/platform/dummy
Register: hello
Register: timer
Register: nsh
Register: sh
Register: getprime
CC:  clock/clock.c Assembler messages:
Error: cannot find default versions of the ISA extension `zicsr'
Error: cannot find default versions of the ISA extension `zifencei'
/tmp/ccyYDT9X.s:4: Error: cannot find default versions of the ISA extension `zicsr'
/tmp/ccyYDT9X.s:4: Error: cannot find default versions of the ISA extension `zifencei'
make[1]: *** [Makefile:64: clock.o] Error 1
make: *** [tools/LibTargets.mk:65: sched/libsched.a] Error 2
lxf@lxf-Inspiron-15-5510:~/nuttxspace/nuttx$ 
```

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-19:
@FlyyingPiggy2020 please try the last master which should be fixed by https://github.com/apache/nuttx/pull/14130.

FlyyingPiggy2020, 2024-10-19:
Thank you for your reply! I have tried the latest branch and confirmed that it compiles successfully. The issue seems to have been resolved. Thanks again for your help and quick response!

closed 2024-10-19 by FlyyingPiggy2020