apache/nuttx #14403: [BUG] CONFIG_LIBCXX doesn't work with cmake

raiden00pl, 2024-10-18:
### Description / Steps to reproduce the issue

It's been broken for a long time but I don't see any issue about it.

```
cmake -B build -DBOARD_CONFIG=sim/libcxxtest -GNinja
cmake --build build
```

```
[872/1512] Building CXX object libs/libxx/CMakeFiles/libcxx.dir/libcxx/src/ios.instantiations.cpp.o
FAILED: libs/libxx/CMakeFiles/libcxx.dir/libcxx/src/ios.instantiations.cpp.o 
/usr/bin/g++ -DLIBCXX_BUILDING_LIBCXXABI -D_LIBCPP_BUILDING_LIBRARY -D__NuttX__ -I/home/raiden00/git/railab/dawn/external/nuttx/libs/libxx/libcxx/src -I/home/raiden00/git/railab/dawn/external/nuttx/build/include/libcxx -I/home/raiden00/git/railab/dawn/external/nuttx/libs/libxx/libcxxabi/include -isystem /home/raiden00/git/railab/dawn/external/nuttx/include -isystem /home/raiden00/git/railab/dawn/external/nuttx/build/include -std=gnu++20 -U_AIX -U_WIN32 -U__APPLE__ -U__FreeBSD__ -U__NetBSD__ -U__linux__ -U__sun__ -U__unix__ -U__ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__ -g -fomit-frame-pointer -ffunction-sections -fdata-sections -fno-common -fvisibility=hidden -Wall -Wshadow -Wundef -Wno-attributes -Wno-unknown-pragmas -std=gnu++20 -nostdinc++ -D__GLIBCXX__ -D_LIBCPP_DISABLE_AVAILABILITY -fmacro-prefix-map=/home/raiden00/git/railab/dawn/external/nuttx= -fmacro-prefix-map=/home/raiden00/git/railab/dawn/external/apps= -fmacro-prefix-map=/home/raiden00/git/railab/dawn/external/nuttx/boards/sim/sim/sim= -fmacro-prefix-map=/home/raiden00/git/railab/dawn/external/nuttx/arch/sim/src/sim= -Wno-deprecated-declarations -Wno-shadow -Wno-sign-compare -MD -MT libs/libxx/CMakeFiles/libcxx.dir/libcxx/src/ios.instantiations.cpp.o -MF libs/libxx/CMakeFiles/libcxx.dir/libcxx/src/ios.instantiations.cpp.o.d -o libs/libxx/CMakeFiles/libcxx.dir/libcxx/src/ios.instantiations.cpp.o -c /home/raiden00/git/railab/dawn/external/nuttx/libs/libxx/libcxx/src/ios.instantiations.cpp
In file included from /home/raiden00/git/railab/dawn/external/nuttx/build/include/libcxx/__filesystem/filesystem_error.h:15,
                 from /home/raiden00/git/railab/dawn/external/nuttx/build/include/libcxx/__filesystem/directory_entry.h:20,
                 from /home/raiden00/git/railab/dawn/external/nuttx/build/include/libcxx/filesystem:539,
                 from /home/raiden00/git/railab/dawn/external/nuttx/build/include/libcxx/fstream:192,
                 from /home/raiden00/git/railab/dawn/external/nuttx/libs/libxx/libcxx/src/ios.instantiations.cpp:10:
/home/raiden00/git/railab/dawn/external/nuttx/build/include/libcxx/__filesystem/path.h: In instantiation of ‘std::__1::__fs::filesystem::path::_EnableIfPathable<_Source> std::__1::__fs::filesystem::path::append(const _Source&) [with _Source = std::__1::basic_string<char>]’:
/home/raiden00/git/railab/dawn/external/nuttx/build/include/libcxx/__filesystem/path.h:623:30: error: use of built-in trait ‘__remove_pointer(typename std::__1::decay<_Tp>::type)’ in function signature; use library traits instead
  623 |   _EnableIfPathable<_Source> append(const _Source& __src) {
      |                              ^~~~~~
[885/1512] Building CXX object libs/libxx/CMakeFiles/libcxx.dir/libcxx/src/locale.cpp.o
ninja: build stopped: subcommand failed.
```

it can be fixed with this changes https://github.com/apache/nuttx/pull/12150 but this PR is closed and now it's hard to rebase these changes with master

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-19:
@xuxin930 @cuiziwei1 please look at this issue.

xuxin930, 2024-10-20:
hi @raiden00pl what version of toolchain are you using?

raiden00pl, 2024-10-20:
I see now that I pasted the wrong error. The error above is for gcc 14, but there is something wrong also for arm gcc 13 (warnings, but no error).  Please look at https://github.com/apache/nuttx/issues/14412
I modified this issue to make it clear that it is for gcc 14

xuxin930, 2024-10-20:
> 我现在发现我粘贴了错误的错误。上面的错误是针对 gcc 14 的，但 arm gcc 13 也出现了问题（警告，但没有错误）。请查看#14412 我修改了这个问题，以明确它适用于 gcc 14

@raiden00pl thanks, I'll reproduce the issue and fix it.

raiden00pl, 2025-02-20:
fixed with https://github.com/apache/nuttx/pull/15865

closed 2025-02-20 by raiden00pl