apache/nuttx #14413: [BUG] environmental variables doesn't work for cmake build

raiden00pl, 2024-10-20:
### Description / Steps to reproduce the issue

when I pass compiler versions as environmental variable it doesn't work:

```
CXX=g++-13 CC=gcc-13 cmake -B build -DBOARD_CONFIG=sim/libcxxtest -GNinja
-- Initializing NuttX                                                                                       
  Select HOST_LINUX=y
  Select HOST_X86_64=y
--   CMake   3.29.5
--   Ninja   1.12.1
--   Board:  sim
--   Config: libcxxtest
--   Appdir: /home/raiden00/git/RTOS/nuttx/apps
-- The C compiler identification is GNU 14.1.1
-- The CXX compiler identification is GNU 14.1.1
```

should be:

```
CXX=g++-13 CC=gcc-13 cmake -B build -DBOARD_CONFIG=sim/libcxxtest -GNinja
 Select HOST_LINUX=y                                                                                        
 Select HOST_X86_64=y
-- Initializing NuttX
--   Board:  sim
--   Config: libcxxtest
--   Appdir: /home/raiden00/git/RTOS/nuttx/apps
-- The C compiler identification is GNU 13.3.0
-- The CXX compiler identification is GNU 13.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc-13
```

This was broken recently.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

xuxin930, 2024-10-21:
https://github.com/apache/nuttx/pull/14427

@raiden00pl Let's not find_program in sim arch

simbit18, 2024-10-21:
@raiden00pl  or see if this can work

cmake -B build -GNinja -DBOARD_CONFIG=sim:libcxxtest -DCMAKE_CXX_COMPILER=`which g++-13` -DCMAKE_C_COMPILER=`which gcc-13`

raiden00pl, 2024-10-21:
#14427 fixes the problem, thanks

closed 2024-10-21 by xiaoxiang781216