apache/nuttx #14418: [BUG] debug log "lost" carrier return in master

FelipeMdeO, 2024-10-20:
### Description / Steps to reproduce the issue

Hello All,

Using realeases/12.7 I am able to see debug output properly:
[image]

But when I move to master branch I am losting carrier return after printfs:
[image]

I am using esp32c6-devkitc:spi config.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.0

### NuttX Version

Master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Kernel]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-21:
is the master or 12.7 branch?

anchao, 2024-10-21:
Regression of https://github.com/apache/nuttx/pull/14362  ?

yangsong8-ai, 2024-10-21:
Please try to set CONFIG_SYSLOG_CRLF=y

closed 2024-10-21 by raiden00pl