apache/nuttx #14467: [BUG]  kernel mode gets stuck at the system call layer during backtrace to user mode

gneworld, 2024-10-23:
### Description / Steps to reproduce the issue

build system in protect mode or kernel mode

1. make a breakpoint at function uart_read
2. type any character in the nsh shell

As you can see, the back trace process is stuck at dispatch_syscall function

[image]

However, xtensa can handle this case correctly.

[image]

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

ubuntu 18.04 x64

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Debugging]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-10-25:
@anchao @xiaoxiang781216 any idea?

xiaoxiang781216, 2024-10-26:
xtensa has the special code to make gdb pass through syscall boundary, @tmedicci does extensa trick apply to arm arch too?