apache/nuttx #14484: [HELP] How to get the more status when NuttX is running

lzc1001, 2024-10-23:
### Description

Hi everyone, 
I have boot NuttX on FVP multi-core, and those cores are armv8a arch, but I don't know how to get the status of each core. 
Thank you every much!

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-24:
which state do you want to get?

lzc1001, 2024-10-24:
@xiaoxiang781216 I want to get each core ID by /proc/cpuinfo on FVP A510, But the related code isn't implemented for ARMV8A. I want get these information to verify multi-core are both active. I am also trying to write a simple multi-task code for verifing this.

lzc1001, 2024-10-24:
@xiaoxiang781216 hhh, I have run apps/testing/smp_main.c to verify each core is activity. Thank you very much, I wonder if there's another way to check or prove this?😄

xiaoxiang781216, 2024-10-24:
ps could also show multiple idle thread, and testing/cpuload could simulate the loading on the specific cpu core.

lzc1001, 2024-10-24:
@xiaoxiang781216 Thank you very much!

closed 2024-10-24 by lzc1001