apache/nuttx #14513: [BUG] When building `nuttx-apps`, CI Workflow checks out the Incorrect Branch of `nuttx` repo

lupyuen, 2024-10-26:
### Description / Steps to reproduce the issue

1. When compiling the Release Branch of `nuttx-apps`, our CI Workflow `nuttx-apps/build.yml` checks out the Master Branch of `nuttx` repo, which is incorrect: https://github.com/apache/nuttx-apps/pull/2770#event-14876028222
   > the CI does not use the right branch
   > modified: boards/arm/mps/mps3-an547/configs/ap/defconfig
   > modified: boards/arm/mps/mps3-an547/configs/bl/defconfig
   > we do not have those configs in releases/12.7

2. This comes from the arm-05 log: https://github.com/apache/nuttx-apps/actions/runs/11518661640/job/32069572079#step:7:96
```text
   Configuration/Tool: mps3-an547/bl,CONFIG_ARM_TOOLCHAIN_GNU_EABI
   2024-10-25 14:11:04
     Normalize mps3-an547/bl
   83d82
   < # CONFIG_NSH_DISABLE_WAIT is not set
   Saving the new configuration file
   On branch master
   Your branch is up to date with 'origin/master'.

   Changes not staged for commit:
     (use "git add <file>..." to update what will be committed)
     (use "git restore <file>..." to discard changes in working directory)
	   modified:   boards/arm/mps/mps3-an547/configs/bl/defconfig
```

3. This happens because `mps3-an547:bl` doesn't exist in the `releases/12.7` branch. [(It exists in `master`)](https://github.com/apache/nuttx/blob/master/boards/arm/mps/mps3-an547/configs/bl/defconfig)

4. Fetch-Source Log shows that CI is indeed checking out the Master Branch of `nuttx` repo (instead of `releases/12.7`): https://github.com/apache/nuttx-apps/actions/runs/11518661640/job/32069568370#step:3:66
   ```text
   /usr/bin/git checkout --progress --force -B master refs/remotes/origin/master
   ```

5. Which might be caused by a typo in `nuttx-apps/build.yml`: https://github.com/apache/nuttx-apps/blob/master/.github/workflows/build.yml#L92
   ```yaml
         - name: Checkout nuttx repo
           uses: actions/checkout@v4
           with:
             repository: apache/nuttx
             ref: ${{ steps.gittargets.outputs.os_ref }}
   ```

6. The workflow above says we are checking out Branch `os_ref`. But `os_ref` is undefined! It's actually named `name: https://github.com/apache/nuttx-apps/blob/master/.github/workflows/build.yml#L85
   ```yaml
     Fetch-Source:
       runs-on: ubuntu-latest
       steps:
         - name: Determine Target Branches
             ...
             echo "name=$OS_REF" >> $GITHUB_OUTPUT
             echo "app_ref=$APPS_REF" >> $GITHUB_OUTPUT
   ```

7. I propose to change `name` to `os_ref`:
   ```yaml
             echo "os_ref=$OS_REF" >> $GITHUB_OUTPUT
             echo "app_ref=$APPS_REF" >> $GITHUB_OUTPUT
   ```

8. Tracking back to the original code: It seems plausible that there's a typo: https://github.com/apache/nuttx/pull/9289/files
   ```yaml
             echo ::set-output name=os_ref::$OS_REF
             echo ::set-output name=apps_ref::$APPS_REF
   ```
   Suspiciously became:
   ```yaml
             echo "name=$OS_REF" >> $GITHUB_OUTPUT
             echo "app_ref=$APPS_REF" >> $GITHUB_OUTPUT
   ```

9. Note the `apps_ref` vs `app_ref`. I patched it to `apps_ref` in
https://github.com/apache/nuttx/pull/13806

Hi @simbit18: Do you agree with my findings? Do you think it will cause issues in the NuttX Repo when we change `name` to `os_ref`? Thanks!

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

CI Workflow

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Other]

### Verification

- [X] I have verified before submitting the report.

lupyuen, 2024-10-26:
Update: We should never apply this fix to `releases/12.7` of `nuttx-apps`. It will fail because the Newer CI Jobs don't exist in `releases/12.7` of `nuttx` repo: `arm64_01.dat`, `x86_64-01.dat`, `risc-v-03.dat`, etc
```text
+ /github/workspace/sources/nuttx/tools/testbuild.sh -A -N -R -S -j 4 -e '-Wno-cpp -Werror' testlist/arm64-01.dat
ERROR: No readable file exists at testlist/arm64-01.dat
```
https://github.com/lupyuen5/label-nuttx-apps/actions/runs/11528123257/job/32094855996#step:7:81

In Future: We should always backport the Latest CI Jobs `*.dat` to the `releases` branch. Right now things a little messy because we're cutting GitHub Runners: https://github.com/apache/nuttx/issues/14376

simbit18, 2024-10-28:
Hi @lupyuen  yes, I think so too that there is a typo due to this change github actions
https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

maybe after coming out of the 30/10 tunnel it would be better to create a workflow **release** with everything (docker, macos, msys2 and msvc) 
build_dev.yml -> development
release.yml -> release
what do you think ?

lupyuen, 2024-10-28:
Thanks @simbit18! Yep let's reorg the CI Workflow after 30 Oct. I'm keen to hear your ideas, maybe you can share an outline of the proposed workflow? Thanks :-)

closed 2024-11-04 by xiaoxiang781216