apache/nuttx #14514: [BUG] hardware timers broken for ARM devices

raiden00pl, 2024-10-26:
### Description / Steps to reproduce the issue

Hardware timer example (`apps/example/timer`) is broken for some of ARM devices (probably for all ARM chips, but I can't check all of them).
What I have verified and is broken: all Nordic chips and all STM32 chips.

Broken with this PR: https://github.com/apache/nuttx/pull/13606

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch Linux

### NuttX Version

nuttx-12.7.0-RC1, master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Drivers]

### Verification

- [X] I have verified before submitting the report.

jasonbu, 2024-11-04:
hi @raiden00pl,
looks like in #7494, we removed the 
```C
        if (lower->ops->getstatus) /* Optional */
          {
...
          }
        else
          {
            ret = -ENOSYS;
          }
```
caused the DEBUG_ASSERT in 
timer_getstatus
```C
  DEBUGASSERT(lower->ops->tick_getstatus);
```
maybe we can keep compatible by return -ENOSYS, if getstatus and tick_getstatus method are both NULL,

if you have more information before which commit the behavior is **worked as expect**, please letme know.
because according to the git log, even with code before #7494, timer demo will also fail, only with no panic.

jasonbu, 2024-11-04:
please review if the #14619 can match your requirements.

raiden00pl, 2024-11-04:
@jasonbu hi, this doesn't fix the problem. NRF52 implements `getstatus` and the timer examples keeps crashing.
For H7 you can just comment out `timer_status` in timer example (it does nothing useful) and it still crashes.

In both cases https://github.com/apache/nuttx/pull/13606 is guilty (works without this change, crashes with it).

raiden00pl, 2024-11-04:
the last working commit is f221c9ecb4 (for H7 you have to remove `timer_status` logic from example, for NRF52 it's not required)

jasonbu, 2024-11-04:
> the last working commit is [f221c9e](https://github.com/apache/nuttx/commit/f221c9ecb436a32ecd14f54053d241722a1816a9) (for H7 you have to remove `timer_status` logic from example, for NRF52 it's not required)

I tried with stm32h7 and return directly in timer_status.
looks like also a sig action happend in timer handler.

dump_stack in up_schedule_sigaction can get backtrace of.

```bash
up_schedule_sigaction arch/arm/src/armv7-m/arm_schedulesigaction.c:104 (discriminator 1)
nxsig_queue_action sched/signal/sig_dispatch.c:199
nxsig_tcbdispatch sched/signal/sig_dispatch.c:540
group_signal_handler sched/group/group_signal.c:155
group_foreachchild sched/group/group_foreachchild.c:78
group_signal sched/group/group_signal.c:228
nxsig_dispatch sched/signal/sig_dispatch.c:735
nxsig_notification sched/signal/sig_notification.c:147
timer_notifier drivers/timers/timer.c:127
stm32_timer_handler arch/arm/src/chip/stm32_tim_lowerhalf.c:272
irq_dispatch sched/irq/irq_dispatch.c:148
exception_direct arch/arm/src/armv7-m/arm_doirq.c:49
```

if it's urgent requirement, you can do 
```C
const void * const _vectors[] locate_data(".vectors")
                              aligned_data(VECTAB_ALIGN) =
{
  /* Initial stack */

  IDLE_STACK,

  /* Reset exception handler */

  start,

  /* Vectors 2 - n point directly at the generic handler */

#if 0
  [2 ... NVIC_IRQ_PENDSV] = &exception_common,
  [(NVIC_IRQ_PENDSV + 1) ... (15 + ARMV7M_PERIPHERAL_INTERRUPTS)]
                          = &exception_direct
#else
  [2 ... ARMV7M_PERIPHERAL_INTERRUPTS] = &exception_common,
#endif
};
```
in arch/arm/src/armv7-m/arm_vectors.c

and we shall talk about how to cover these kind of usage.

CC @GUIDINGLI @xiaoxiang781216

raiden00pl, 2024-11-20:
@jasonbu @GUIDINGLI @xiaoxiang781216 do you plan to fix this issue ?

jasonbu, 2024-12-04:
> @jasonbu @GUIDINGLI @xiaoxiang781216 do you plan to fix this issue ?

in-processing, will provide a draft patch after my local qemu mps2 test as soon as possible.
use pendsv to deliver the signal.

raiden00pl, 2024-12-05:
still broken on NRF52

jasonbu, 2024-12-05:
hi @raiden00pl  are you still break in same case?  in my case break on common/arm_exit.c
and I'm looking at what's happening.

jasonbu, 2024-12-06:
@hujun260 please review this issue,
my local enviroment also reproduce the issue. but if open CONFIG_ARCH_IRQPRIO, the case can passed(fix caused by https://github.com/apache/nuttx/commit/2b22ee03d6a44f82c94a557e5c0c41932718c5d1 as a prefix).

raiden00pl, 2024-12-09:
@hujun260 @jasonbu  `armv6-m` is still broken

raiden00pl, 2024-12-09:
@hujun260  I checked again and everything is OK. Sorry for the confusion, something was wrong on my end. Thanks for fixing it

closed 2024-12-09 by raiden00pl