apache/nuttx #14531: [BUG] TCP will release when FTP Client want to connect in FTPD example

SmartIotMaker, 2024-10-28:
### Description / Steps to reproduce the issue

1. Test **ESP32-S3-DevKitC-1 v1.0**: esp32s3-devkit:sta_softap with CONFIG_EXAMPLES_FTPD=y and CONFIG_NETUTILS_FTPD.

2. nuttx base on **master** and last commit is `https://github.com/apache/nuttx/commit/204871513486e940fbb29a9a524de9b9e063d3e0` 

3. tool chain：

```
xtensa-esp32s3-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/home/john/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/xtensa-esp-elf-gcc
COLLECT_LTO_WRAPPER=/home/john/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../libexec/gcc/xtensa-esp-elf/14.2.0/lto-wrapper
Target: xtensa-esp-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp-elf --exec_prefix=/builds/idf/crosstool-NG/builds/xtensa-esp-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp-elf/xtensa-esp-elf --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp-elf/xtensa-esp-elf --with-native-system-header-dir=/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-14.2.0_20240906' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-libstdcxx-verbose --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 14.2.0 (crosstool-NG esp-14.2.0_20240906)
```

4. then  flash nuttx.bin and config wifi with `wapi psk wlan0 12345678 3;wapi essid wlan0 John 1;renew wlan0`
5. run ftpd with `ftpd_start -4` and connect it with ftp client tools. 
6. ftp client will disconnet after input password(usename:root, password:abc123).
7. dump with WireShark tool and find board release connect after SYN.
8. shell will print `WARNING: Listen canceled while waiting for ACK on port`(line 1287 in tcp_imput.c),this means no any listener in list.

after I modify ftpd_session(handle,5000) to ftpd_session(handle, -1) and connect is ok.
follow is the patch:
```
diff --git a/examples/ftpd/ftpd_main.c b/examples/ftpd/ftpd_main.c
index d1052b50c..208624c1c 100644
--- a/examples/ftpd/ftpd_main.c
+++ b/examples/ftpd/ftpd_main.c
@@ -229,7 +229,7 @@ int ftpd_daemon(int s_argc, char **s_argv)
        * has been started.
        */
 
-      ret = ftpd_session(handle, 5000);
+      ret = ftpd_session(handle, -1);
 
       /* If any interesting happened (i.e., any thing other than a timeout),
        * then report the interesting event.
```
this change make accept4() will immediately call after listen().

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Applications]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-10-28:
@wengzhe please take a look

wengzhe, 2024-10-29:
Hi @SmartIotMaker , have you tried to enable `CONFIG_NET_TCPBACKLOG`? With backlog, the server won't need to accept immediately when client connects.

SmartIotMaker, 2024-10-29:
@wengzhe yes, everything is OK when enable CONFIG_NET_TCPBACKLOG, thank you.
but it is seemingly very strange because only enable log.

wengzhe, 2024-10-29:
@SmartIotMaker Glad to see your problem solved. The "backlog" is not a "log", you may see it as the second parameter of `listen` on a socket, which "[limit the number of outstanding connections in the socket's listen queue](https://pubs.opengroup.org/onlinepubs/9699919799/functions/listen.html)". Without `CONFIG_NET_TCPBACKLOG` enabled, we won't have a listen queue (which means `backlog=0`), so we cannot accept clients asynchronously, which causes your problem.

closed 2024-10-29 by xiaoxiang781216

SmartIotMaker, 2024-10-30:
@wengzhe Thank you very much for your explanation. I got it.