apache/nuttx #14532: crc32 -CRC is not compatible with Linux

shijogeorge24, 2024-10-28:
### Description / Steps to reproduce the issue

I was working with a simple UART communication layer between Linux and NuttX and encountered an issue related to CRC32. When I compared the CRC implementations on Linux and NuttX, I found that they differ.

You can refer to the previous NuttX issue here: https://lists.apache.org/thread/hjw83nb1mwzh0d24oprpt0np7kfxwdrk.

What is the best way to resolve this?

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu23

### NuttX Version

12.6.0 

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: OS Components]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-10-28:
add a new crc32 api like crc8 or crc16 which has many variants.

acassis, 2024-10-28:
Seems like even in different versions of python the CRC32 is different:

https://stackoverflow.com/questions/30092226/calculate-crc32-correctly-with-python

And also when the file is bigger than 2GB:

https://github.com/python/cpython/issues/105967

@xiaoxiang781216 I think we need to convert our default crc32 to become compatible with Linux and move the current to a different name and use it on existing subsystems that requires the current implementation.

What do you think?

xiaoxiang781216, 2024-10-28:
I am fine with this change since the compatibility with other popular POSIX OS is always my top priority.

jlaitine, 2024-10-29:
Is the difference just init value or polynomial itself? Maybe keep the default crc32, there is some reason why exactly that variant has selected (I don't know what), but blindly chaning the defaults would break it for people already using it.

There is no one "correct crc32", but many variants. If the difference is just init value, you can use "crc32part" directly to calculate it starting with whatever init value you want.

acassis, 2024-10-29:
Hi @jlaitine actually it is not a blindly change, many people faced this issue in the past, but  I agree it could break other people code and probably this modification will require changing the NuttX major number. As you said, there is not a "correct crc32", but it is important to select a default variant compatible with Linux.