apache/nuttx #14550: regarding folder common/espressif

yf13, 2024-10-29:
Currently we have `arch/risc-v/src/commmon/espressif` which holds `espressif` specific things only.

Since `arch/risc-v/src/common` reads like things common to all RiscV chip, I am wondering if it can be moved out of common folder of all RiscV chips? probably it can stay in a folder like `arch/risc-v/src/espressif`?

This will be more in line what we have done for `arch/arm/src/common`.

xiaoxiang781216, 2024-10-30:
Yes, I think so.

yamt, 2024-10-30:
i guess whatever decision made about it would apply to arch/xtensa/src/common/espressif too.

yf13, 2024-10-30:
@tmedicci can we have your opinion? :-)

tmedicci, 2024-10-30:
I prefer to keep it in `common` folder because it's common to all Espressif SoCs.

We used to have something like that and named the RISC-V-based SoCs just as `espressif`. This old organization was very confusing and we always had people asking about the SoCs. Then, we revert it to make it easier to identify what is a common code source and what is specific to a chip only.
See, `nuttx/arch/risc-v/src` contains chips (and `espressif` isn't a chip). These chips may (or not) use a common code base (the code at `arch/risc-v/src/commmon/espressif`) or implement their own version of some peripheral/driver. The current organization allows doing that more straightforward. We thought a lot about the location of these folders trying to keep a consistency between existing and yet-to-be-supported SoCs on NuttX.

yf13, 2024-10-31:
@tmedicci, thanks for sharing the history.

Can you teach why `arch/risc-v/src/espressif` is confusing people? I can't see it because specific names like `arch/risc-v/src/esp32c3` will attract me at the first place.

yf13, 2024-11-01:
Thank you @tmedicci for the explanations.

Seems you are assuming that subfolders of `arch/risc-v/src/` must be chip specific but `espressif` isn't chip specific.

However, I think the [directory structure](https://nuttx.apache.org/docs/latest/quickstart/organization.html) design just says that chip specific folder should be subfolder of `arch/risc-v/src`, but it doesn't say that subfolders of `arch/risc-v/src` must all be chip specific.

Taking the `stm32` example, it depends on `armv7-m` which further depends on `common`. Here both `armv7-m` and `common` are subfolders in `arch/arm/src` and are not chip specific.

tmedicci, 2024-11-01:
> Taking the `stm32` example, it depends on `armv7-m` which further depends on `common`. Here both `armv7-m` and `common` are subfolders in `arch/arm/src` and are not chip specific.

Yes, exactly! This is confusing: if `armv7-m` is not a chip by itself, it should be moved to `common` because it provides source code to more than one chip of different vendors (I supposed about the various vendors). It isn't only about the path, but Kconfig and build-related files need to be adjusted to include "common" files, some of the "common" source files are in a sub-directory called `common`, others are in the same directory level. It isn't that easy to figure out that...  See, NuttX is growing and people are interested in supporting more devices: we need to create a clear path for our sources. We had that for RISC-V and Xtensa-based devices: we just kept using the same logic.