apache/nuttx #14551: [HELP] STM32 Macro Naming Conventions

stbenn, 2024-10-29:
### Description

There seems to be two naming conventions when defining macros for STM32 boards:

1. Identifiers typically omit chip type, and start with `STM32_`
    - An example would be the STM32H7. In `arch/arm/src/stm32h7/hardware/`, and the `board.h` file for the nucleo-h743zi (`boards/arm/stm32h7/nucleo-h743zi/include/board.h`
1. Identifiers typically include chip type, and start with `STM32XX_`
    - An example would be the STM32L5. In `arch/arm/src/stm32l5/hardware/`, and the `board.h` file for nucleo-l552ze (`boards/arm/stm32l5/nucleo-l552ze/inclue/board.h` 

I cannot figure out why some chips use the first style, and some use the second. 

Are the differences between these naming conventions purely stylistic, or are there specific reasons to use one convention instead of the other? 

When adding support for a new STM32 chip family, which naming convention would be preferred?

### Verification

- [X] I have verified before submitting the report.

raiden00pl, 2024-10-30:
Please use the shorter version: `STM32_`. The longer version comes from `stm32l4` port and the ports using it are most likely based on `stm32l4` port (or any of its later derivatives).
The shorter version is easier to maintain, helps when we have to update all stm32 chips at once and porting between stm32 boards is then easier.
I think `STM32XX_` should be modifed to `STM32_` someday.

stbenn, 2024-10-30:
Should function and file names follow this shorter version as well then?

closed 2024-12-03 by stbenn