apache/nuttx #14558: [FEATURE] Build Dashboard for NuttX Build Farm

lupyuen, 2024-10-30:
### Is your feature request related to a problem? Please describe.

More details coming! Try it here: https://nuttx-dashboard.org

[image]

### Describe the solution you'd like

TODO

### Verification

- [X] I have verified before submitting the report.

lupyuen, 2024-11-01:
### GitHub Actions CI Logs are now inside NuttX Dashboard

[Our NuttX Dashboard](https://nuttx-dashboard.org) now ingests the CI Build Logs from GitHub Actions! (Like an Amoeba)
- That's why our Dashboard looks Red, because of errors in the CI Build (pic below)
- When we click a Red Box, it jumps to the GitHub Actions Log (showing the error)

How It Happens:
- Daily at 00:00 UTC and 12:00 UTC: I click on [NuttX Mirror Repo](https://github.com/NuttX/nuttx) > Sync Fork > Discard Commits
- Run this script to Enable the macOS and Windows Builds: [enable-macos-windows.sh](https://github.com/lupyuen/nuttx-release/blob/main/enable-macos-windows.sh)
- Wait 2.5 Hours for the build to complete
- Then run this script to Ingest the GitHub Actions Logs into NuttX Dashboard: [ingest-nuttx-builds/github.sh](https://github.com/lupyuen/ingest-nuttx-builds/blob/main/github.sh)
- [Source Code for Log Parsing is all here](https://github.com/lupyuen/ingest-nuttx-builds)

What about ingesting macOS / msys2 / msvc Logs? Right now they [show tooooo many](https://github.com/NuttX/nuttx/actions/runs/11621721723/job/32366030243#step:7:30302) ... [warningses](https://github.com/NuttX/nuttx/actions/runs/11621721723/job/32366029366#step:9:578). We should probably fix the warnings before ingesting the logs into NuttX Dashboard.

[image]

cederom, 2024-11-03:
Very cool congratz @lupyuen  :-) And its already working :-) I need to buy a disk to a laptop that I can use to join The NuttX Farmers :D

lupyuen, 2024-11-03:
Thanks @cederom! Yep hope to have your laptop in our NuttX Build Farm real soon :-)

cederom, 2024-11-03:
What minimal disk is recommended 500GB or 1TB?

lupyuen, 2024-11-03:
@cederom 500 GB should be more than sufficient. I'm using 51 GB right now :-)
https://gist.github.com/nuttxpr/34b8fa96537acdf7400b87d0e6794cf1#file-ci-arm-14-log-L905

lvanasse, 2024-11-05:
@lupyuen by any chance, do you have a Dockerfile for the whole setup (I know, I've been talking about docker for a while (it'll be after the doc migration;)))

Reason why I'm asking, is that I have an Unraid server at home, which I host my media. And I was thinking maybe it would be possible to simply docker another docker on it.

lupyuen, 2024-11-05:
@lvanasse Yep the Dockerfile is here: https://github.com/apache/nuttx/blob/master/tools/ci/docker/linux/Dockerfile

If you need the Docker Image, I can export it for you :-)

simbit18, 2024-11-06:
Hi @lupyuen this PR #14669 14654 removed the warnings now you can enter msys2 logs in NuttX Dashboard.

lupyuen, 2024-11-06:
@simbit18 Awesome! Lemme work on it...

lupyuen, 2024-11-06:
Hi @simbit18: [msys2 Builds](https://github.com/lupyuen/ingest-nuttx-builds/commit/22e745ee9865ff4de2f58c8e0f143c4c1d8157b2) are now in nuttx-dashboard.org :-)

[image]

simbit18, 2024-11-07:
Thank you @lupyuen

lvanasse, 2024-11-09:
@lupyuen I'm setting up a server to run the build and to potentially add it to the build farm :). I was wondering if you just put the `run-ci.sh` inside a cronjob to be able to periodically run the build?

Also, I think you'll need by Gist ID? It says in your article, that it's better to have another account for security? I'm not sure if I understand why you might need to do that. 

And also, is there a target group that would be best for me to set the CI to run on?

Cheers :)

lupyuen, 2024-11-09:
Thank you so much @lvanasse!
- run-ci.sh inside a cronjob: Might work? Assuming that it can handle the sudo.
- Originally I assumed the Build PC would be compiling NuttX 24 x 7 (because New PRs will be merged all the time), but it's perfectly OK to run it sometimes
- Creating a GitHub Account: The NuttX Dashboard will be crawling all your Gists. If we're comfortable letting a bot crawl through your Gists, then it's OK :-)
- Also the `gh` command will be authenticated using your GitHub Account, which means it has all repo access rights, and it might be too powerful. So it depends if you're comfortable with that :-)

lvanasse, 2024-11-09:
Sounds good! Then if you are not running the script in a cronjob how are you automating the process?

lupyuen, 2024-11-09:
@lvanasse It runs on my Ubuntu Desktop all day all night, I'm not using the PC for anything else :-)

lvanasse, 2024-11-09:
@lupyuen, sorry just so I understand correctly, so you do launch the script manually each time it finished?

lupyuen, 2024-11-09:
@lvanasse Oops sorry, please use this script, it will run all jobs forever:

https://lupyuen.github.io/articles/ci2#build-nuttx-for-all-target-groups

```bash
## Download the scripts
git clone https://github.com/lupyuen/nuttx-release
cd nuttx-release

## Login to GitHub in Headless Mode
sudo apt install gh
sudo gh auth login

## (1) What Account: "GitHub.com"
## (2) Preferred Protocol: "HTTPS"
## (3) Authenticate GitHub CLI: "Login with a web browser"
## (4) Copy the One-Time Code, press Enter
## (5) Press "q" to quit the Text Browser that appears
## (6) Switch to Firefox Browser and load https://github.com/login/device
## (7) Enter the One-Time Code. GitHub Login will proceed.
## See https://stackoverflow.com/questions/78890002/how-to-do-gh-auth-login-when-run-in-headless-mode

## Run the Build Job forever: arm-01 ... arm-14
sudo ./run-ci.sh
```

lvanasse, 2024-11-09:
Ahh ok sorry, I didn't read the whole script properly, and didn't see the infinite loop. 

Thank you for the clarification, I'm setting up the server at the moment and I'll let you know my Gist ID after everything is running smoothly :).

lvanasse, 2024-11-09:
Things seems to be running smoothly :): https://gist.github.com/lvanasse/74976165ac7605a7376a6c026ff020ee

If you want to add my ID to the Nuttx Build Farm, I would be honoured :)!

After it ran a first time through all the target groups, I might just stop it and set it up so I can forget about it and let it run on my server.

lupyuen, 2024-11-09:
Super Awesome @lvanasse! Your PC is now in our NuttX Build Farm, thank you so much! :-)

https://nuttx-dashboard.org/d/fe2q876wubc3kc/nuttx-build-history?from=now-2d&to=now&timezone=browser&var-arch=$__all&var-subarch=$__all&var-board=$__all&var-config=$__all&var-group=$__all&var-Filters=user%7C%3D%7Clvanasse

[image]

Here's how I added your PC to our Build Farm. macOS will join us real soon :-)
https://github.com/lupyuen/ingest-nuttx-builds/commit/4cbaf145a5799cf118df78724a00723d833df71b

lvanasse, 2024-11-09:
Awesome! Thank you for everything @lupyuen!

lvanasse, 2024-11-09:
@lupyuen you might have seen some funniness in my gists. I had a couple of issue with my unraid VM, but things should be smooth from now on :).

lupyuen, 2024-11-09:
@lvanasse haha cool no worries :-)

lupyuen, 2024-11-24:
### [Article] Continuous Integration Dashboard for NuttX (Prometheus and Grafana)

Last article we spoke about the (Twice) Daily Builds for NuttX. Today we talk about Monitoring the Daily Builds (also the NuttX Build Farm) with our new NuttX Dashboard:

(1) How we created our Dashboard with Grafana (open-source)

(2) Pulling the Build Data from Prometheus (also open-source)

(3) Which is populated by Pushgateway (staging database)

(4) Integrated with our Community Build Farm and GitHub Actions

(5) Why do all this? Because we can’t afford to run Complete CI Checks on Every Pull Request!

(6) We expect some breakage, and NuttX Dashboard will help with the fixing

Check out the article: https://lupyuen.github.io/articles/ci4

closed 2024-11-24 by lupyuen