apache/nuttx #14564: [FEATURE] Unified the argv parameters of pthread_create and task_creat

zyfeier, 2024-10-30:
### Is your feature request related to a problem? Please describe.

Currently, the argv types of pthread_create and task_create are different, which can cause confusion during use.

### Describe the solution you'd like

If the argv parameter of task_create can also be replaced with the void * type, it would achieve consistency in form and eliminate the need to dynamically calculate and store based on the number of argv arguments.

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-11-01:
some related issue: https://github.com/apache/nuttx/issues/1108

xiaoxiang781216, 2024-11-01:
@patacongo could you review this suggestion which make kernel thread decouple from task creation code path.