apache/nuttx #14639: [HELP] SPI Slave example stopped work after nuttx example changes

FelipeMdeO, 2024-11-04:
### Description

Hello All,

SPI Slave example stopped work after commit 761ee819568a52f1934ccf17bae20fe9816ebdec. @yamt  do you have any idea why its happen?

I am running in esp32c6 mcu, spi example can be found in: examples/spislv_test

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-11-04:
Guys we need real HW testing! We cannot break the system this way every time a new commit is submitted.

@tmedicci do you have SPI Slave implemented in your HW CI testing?

eren-terzioglu, 2024-11-04:
Hi @acassis,
We don't have a hardware test for SPI Slave on CI.

acassis, 2024-11-04:
Thank you @eren-terzioglu for confirming!
I think this could be a nice test to avoid regression like this

FelipeMdeO, 2024-11-04:
Guys, If you want I can supply hardware (esp32c6 devkits) and remote machine to run a hardware test. Let me know if I can help in this improvement.

FelipeMdeO, 2024-11-07:
Hello @yamt ,

Do have any idea why changes crashed spi slave feature ?

closed 2024-11-10 by FelipeMdeO