apache/nuttx #14660: [BUG] esp32s3 gdb debug problem with CONFIG_ESPRESSIF_SIMPLE_BOOT

chirping78, 2024-11-06:
### Description / Steps to reproduce the issue

Take `esp32s3-devkit:sotest` as an example, with legacy boot format, the board can be debugged. The steps are:

1. add `CONFIG_ESP32S3_APP_FORMAT_LEGACY=y` to boards/xtensa/esp32s3/esp32s3-devkit/configs/sotest/defconfig
2. build (needs esp-bins of bootloader and partition table)
```
./tools/configure.sh -E esp32s3-devkit:sotest
make -j8 flash ESPTOOL_PORT=/dev/ttyUSB0 ESPTOOL_BINDIR=/path/to/esp-bins/
```
3. launch openocd `openocd -f board/esp32s3-builtin.cfg `
4. gdb debug (it's working as expected)
```
xtensa-esp32s3-elf-gdb nuttx
 target remote :3333
 monitor reset halt
 b nx_start
 c
```
With `CONFIG_ESPRESSIF_SIMPLE_BOOT`, gdb debug is not working.  The steps are:

1.  delete `CONFIG_ESP32S3_APP_FORMAT_LEGACY=y` from boards/xtensa/esp32s3/esp32s3-devkit/configs/sotest/defconfig
2. build
```
./tools/configure.sh -E esp32s3-devkit:sotest
make -j8 flash ESPTOOL_PORT=/dev/ttyUSB0
```
3. launch openocd `openocd -f board/esp32s3-builtin.cfg `
4. gdb debug 
```
xtensa-esp32s3-elf-gdb nuttx
 target remote :3333
 monitor reset halt
 b nx_start
 c
```
After `c`, the program can't continue, gdb shows:
```
(gdb) c
Continuing.
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x42030cec

Command aborted.
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 20.04

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Debugging]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-11-06:
@tmedicci could you look at this problem? We want to optimize xtensa arch with esp32s3.

eren-terzioglu, 2024-11-06:
Hi @chirping78,

Due to simpleboot implementation, please force hardware breakpoints with adding  `-c 'set ESP_FLASH_SIZE 0'` parameter to `openocd` command. In your example openocd command should be `openocd -c 'set ESP_FLASH_SIZE 0' -f board/esp32s3-builtin.cfg`

eren-terzioglu, 2024-11-06:
> Hi @eren-terzioglu, tried to add `-c 'set ESP_FLASH_SIZE 0'`, and it works. But the hardware breakpoints have size limit, currently it seems only support 2 hw breakpoints.
> 
> So if use normal `b` breakpoint, I have to delete oldest ones. Or use `tb` to set temporary breakpoint.
> 
> Is this inconvenience is expected?

It was about (if I remember correctly) place where bootloader resides, so it should be expected behaviour.

closed 2024-11-06 by chirping78