apache/nuttx #14662: [BUG] low level console i/o (up_putc) can interfere interrupt-based uart i/o

yamt, 2024-11-06:
### Description / Steps to reproduce the issue

for some configurations, it's possible for a system to use both of low-level up_putc and the uart (drivers/serial/serial.c) on the console.
for an example, printf(stdout) goes to the uart and syslog() goes to up_putc.

the uart uses a semaphore (dev->xmitsem) interlocked with the kernel critical section to deal with interrupts.

although how up_putc is implemented is device-specific, it usually disables interrupts, performs polling-based i/o, and then restores the interrupts.

unfortunately, with SMP, the latter sometimes can "consume" the hardware events which would otherwise have triggered tx interrupts, which the former might be waiting for. if it happens, the former can block forever.

i was mainly looking at the esp32 implementation. but other implementations look similar.

an obvious fix would be to wrap up_putc with the critical section. but i'm not feeling too happy with the approach.

any suggestions?

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 14.7

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Drivers]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-11-06:
@patacongo what do you think? I know this implementation exists since initial NuttX versions

patacongo, 2024-11-07:
> @patacongo what do you think? I know this implementation exists since initial NuttX versions

up_putc was only added originally to get some very early output to assist in CPU bringup and analyzing assertions.  That was about 17 or more years ago.  It normally just printed things like ABCDEF to let you know where the power-up reset logic failed..

For some of the early CPUs, I had only LEDs and low level serial output to perform the entire bring-up.  It is really amazing what you can do with very little if you have to.

up_putc was used in various other, less appropriate ways since...  Such as getting interrupt level serial output.

Using serial output to support CPU bring-up with serial output is not so important with today's modern, cheap debuggers.  Perhaps up_putc() could even just be removed.

yamt, 2024-11-07:
> up_putc was only added originally to get some very early output to assist in CPU bringup and analyzing assertions. That was about 17 or more years ago. It normally just printed things like ABCDEF to [...]

it's still useful for early in the boot, where i guess it's fine. (maybe until we boot secondary CPUs.)

i guess we can avoid most of problematic cases by removing CONFIG_SYSLOG_DEFAULT. how do you think?

acassis, 2024-11-07:
@yamt I agree! It still useful and we need to keep it. No idea why Syslog is using it by default, it could be an option, instead of default output channel

yamt, 2024-11-19:
> unfortunately, with SMP,

i guess this was not really SMP-specific.
consider that:
1. a thread calls syslog, which ends up with calling up_putc. it's preempted in the middle of up_putc.
2. another thread calls uart_write. the tx buffer happens to be full and it blocks.
3. the first thread is scheduled again and breaks the expectation of the blocking thread. (eg. by "restore"ing the tx interrupt status.)

https://github.com/apache/nuttx/pull/14761 should be enough to fix the case as well.

yamt, 2024-11-19:
> but is uart_write protected by the critical section too?

i'm not sure what you meant. but a literal answer is "yes".

casaroli, 2026-11-08:
Possibly fixed by #14761.

closed 2026-11-08 by casaroli