apache/nuttx #14680: [FEATURE] CI Test should always terminate after 1 hour

lupyuen, 2024-11-07:
### Is your feature request related to a problem? Please describe.

CI Test will sometimes run for 4 hours (or longer):
- https://github.com/apache/nuttx/pull/14667#issuecomment-2461197521

This will increase our usage of GitHub Runners. Which may overrun the GitHub Actions Budget allocated by ASF.

### Describe the solution you'd like

CI Test should complete within 1 hour. It should gracefully terminate itself (and report an error) if the runtime exceeds 1 hour.

### Describe alternatives you've considered

Right now I'm manually killing all CI Jobs that run over 3 hours.

### Verification

- [X] I have verified before submitting the report.

simbit18, 2024-11-07:
@lupyuen maybe we also need to put a maximum number of minutes for a job to run.

GitHub Actions timeout
 https://docs.github.com/en/actions/writing-workflows/workflow-syntax-for-github-actions#jobsjob_idtimeout-minutes

lupyuen, 2024-11-07:
@simbit18 Yep right now it quits after 6 hours: https://github.com/NuttX/nuttx/actions/runs/11714861244

simbit18, 2024-11-07:
@lupyuen so we should put

jobs:
  build:
    runs-on: ubuntu-latest
    timeout-minutes: 180 #  (3 hours) Increase this timeout value as needed

lupyuen, 2024-11-07:
@simbit18 Hmmm suppose right after CI Test there's another build. If CI Test runs for all 3 hours, then the build after CI Test will never run. So actually I prefer if CI Test could terminate itself (after 1 hour) and let other builds run.

Unless we always park CI Test at the end of the job?

simbit18, 2024-11-07:
`@simbit18 Hmmm suppose right after CI Test there's another build. If CI Test runs for all 3 hours, then the build after CI Test will never run. So actually I prefer if CI Test could terminate itself (after 1 hour) and let other builds run.`
right !!!

```
Describe the solution you'd like
CI Test should complete within 1 hour. It should gracefully terminate itself (and report an error) if the runtime exceeds 1 hour.
```
This in my opinion is the right solution

The GitHub Actions timeout is only for safety and not to fall back into the tunnel https://github.com/apache/nuttx/issues/14376

lupyuen, 2024-11-15:
It's happening again:
- https://github.com/apache/nuttx/issues/14808

lupyuen, 2024-11-16:
Wonder if this will work for GitHub CI? I'm testing it for macOS Build Farm:
https://github.com/lupyuen/nuttx-build-farm/blob/main/run-job-macos.sh#L131-L144
```bash
## If CI Test Hangs: Kill it after 1 hour
( sleep 3600 ; echo Killing pytest... ; pkill -f pytest )&

## Run the CI Job
./cibuild.sh -i -c -A -R testlist/$job.dat
```

lupyuen, 2024-11-16:
Yep this kills the CI Test after 2 hours! (Assuming our jobs are not supposed to exceed 2 hours)

We changed [build.yml](https://github.com/NuttX/nuttx/commit/0fc9e07fbf4da289f7aa544254787611fccb0954#diff-5c3fa597431eda03ac3339ae6bf7f05e1a50d6fc7333679ec38e21b337cb6721R184-R192):
```bash
cd sources/nuttx/tools/ci
if [ "X${{matrix.boards}}" = "Xcodechecker" ]; then
  ./cibuild.sh -c -A -N -R --codechecker testlist/${{matrix.boards}}.dat
else
  ## Inserted this
  ( sleep 7200 ; echo Killing pytest... ; pkill -f pytest )&
  ./cibuild.sh -c -A -N -R -S testlist/${{matrix.boards}}.dat
fi
```
[(See the Build Log)](https://github.com/NuttX/nuttx/actions/runs/11871069649/job/33083208869)

- Which is patched into NuttX Mirror Repo by this script: [enable-macos-windows.sh](https://github.com/lupyuen/nuttx-release/blob/main/enable-macos-windows.sh#L60-L70)
- Also for Ubuntu Build Farm: [run-job.sh](https://github.com/lupyuen/nuttx-release/blob/main/run-job.sh#L33-L38)
- And for macOS Build Farm: [run-job-macos.sh](https://github.com/lupyuen/nuttx-build-farm/blob/main/run-job-macos.sh#L131-L144)