apache/nuttx #14690: [BUG] Undefined GPIO Constants (GPIO_SSI1_CLK, GPIO_SSI1_RX, GPIO_SSI1_TX) in tiva_ssi.c on TM4C123G

angelorisonk, 2024-11-07:
### Description / Steps to reproduce the issue

When enabling SSI1 on the Tiva TM4C123G board in NuttX, the following errors occur during the build:

1. `GPIO_SSI1_CLK`, `GPIO_SSI1_RX`, and `GPIO_SSI1_TX` are reported as undeclared identifiers in `tiva_ssi.c`.
2. The build fails as a result of these undefined constants, even though similar constants (e.g., `GPIO_SSI3_CLK`) are available.

Steps to reproduce:
1. Configure the NuttX project for the Tiva TM4C123G board.
2. Enable SSI1 through `make menuconfig`:
   - Navigate to **Device Drivers** > **SPI Support** > **Enable SSI1**
3. Build the project with `make -j$(nproc)`.
4. Observe the compilation errors related to the undefined `GPIO_SSI1_CLK`, `GPIO_SSI1_RX`, and `GPIO_SSI1_TX`.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Linux Mint 22 

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Other]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-11-07:
Hi @angelorisonk please look how it is implemented for SSI3 at boards/arm/tiva/dk-tm4c129x/include/board.h

Then look at arch/arm/src/tiva/hardware/tm4c/tm4c_pinmap.h to associate the right GPIO_SSI1_CLK_x GPIO_SSI1_RX_x and GPIO_SSI1_TX_x symbol with the GPIO pin you are using.

@hartmannathan since you are more familiar with this chip, is there any other thing to observe?

hartmannathan, 2024-11-08:
> Hi @angelorisonk please look how it is implemented for SSI3 at boards/arm/tiva/dk-tm4c129x/include/board.h
>
>
>
> Then look at arch/arm/src/tiva/hardware/tm4c/tm4c_pinmap.h to associate the right [...]

@acassis I'm writing this from my phone right now but if there is still a problem I can look into it more carefully tomorrow. I think you are correct: when the chip can map a signal to alternative pins, the board needs to choose which one to use by providing the define. For example:

```
#define GPIO_SSI1_CLK GPIO_SSI1_CLK_1
```

Obviously you need to use the correct mapping. To see which pin `GPIO_SSI1_CLK_1` maps to, grep for that name in the arch/arm subtree.

angelorisonk, 2024-11-08:
@hartmannathan  Thank you for your suggestion. However, after adding the lines you recommended, I am still encountering the same issue. I have spent several hours debugging, but the problem persists, and I can't seem to identify the root cause. Any further guidance or insights would be greatly appreciated.

acassis, 2024-11-08:
@angelorisonk please run a "git diff" to show the lines you added. As I said you need to look at the pinmap header file at arch/ and create the definitions at boards/..../include/board.h point to each pin:

#define GPIO_SSI1_CLK GPIO_SSI1_CLK_x
#define GPIO_SSI1_RX GPIO_SSI1_RX_y
#define GPIO_SSI1_TX GPIO_SSI1_TX_z

Just fix x, y and z according to pinmap symbol to the right pin you are using in your board.

angelorisonk, 2024-11-15:
@hartmannathan Thank you so much for your detailed explanation and guidance! 🙏 Your clarification on the pin mappings and the _1 and _2 suffixes really helped me understand the issue better.

I added the following definitions to the nuttxspace/nuttx/boards/arm/tiva/tm4c123g-launchpad/include/board.h file as you suggested:

#define GPIO_SSI1_CLK  GPIO_SSI1_CLK_1 //PD0
#define GPIO_SSI1_RX   GPIO_SSI1_RX_1   //PD2
#define GPIO_SSI1_TX   GPIO_SSI1_TX_1   //PD3 

After updating the file with these mappings, the build errors have been resolved, and the SPI communication is now working as expected. 🎉

I truly appreciate your assistance and the time you took to help me out. Thanks again for your support! 😊

hartmannathan, 2024-11-15:
@angelorisonk glad to have helped!