apache/nuttx #14738: [BUG] lib_outstream_s putc and puts are redundant

yamt, 2024-11-12:
### Description / Steps to reproduce the issue

lib_outstream_s has two seemingly-redundant methods, one for single-char operation and another for multi-chars operation, for no obvious reasons. (well, at least not obvious to me.)
if it's necessary for some reasons, it should have been commented in the code.
otherwise, one of them, maybe the single-char version which is likely slower, should be removed to save code size.

ditto for
* lib_instream_s getc and gets
* syslog_putc and syslog_write
* syslog sc_putc/sc_force and sc_write/sc_write_force
* up_putc and up_nputs

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 14.7

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Api]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-11-12:
puts is added later to improve the performance, it's better to drop putc.

yamt, 2025-01-20:
@anchao what do you think about this? do you want to keep putc for possible optimizations for the 1-byte case?

anchao, 2025-01-20:
> [@anchao](https://github.com/anchao) what do you think about this? do you want to keep putc for possible optimizations for the 1-byte case?

we are on the way:
https://github.com/apache/nuttx/pull/14810

 Every cycle, every machine instruction optimization is a great progress

yamt, 2025-01-20:
> > [@anchao](https://github.com/anchao) what do you think about this? do you want to keep putc for possible optimizations for the 1-byte case?
> 
> we are on the way: [#14810](https://github.com/apache/nuttx/pull/14810)
> 
> Every cycle, every machine instruction optimization is a great progress

you misunderstood my question. i meant the opposite.
i was afraid that you were against this issue because it were possible to optimize the 1-size case better with putc-like interfaces.

anchao, 2025-01-20:
I am looking forward to using uio to adapt to the scenario of large memory copy, and the overhead of a single byte is too much, The scenario of putc is just the opposite. puts will reduce function jump branch and push stack. we should avoid using single-byte assignments If possible.