apache/nuttx #14744: [BUG] olimex-stm32-p407/kmodule: .bss will not fit in region ksram

lupyuen, 2024-11-12:
### Description / Steps to reproduce the issue

`olimex-stm32-p407:kmodule` build is failing in our NuttX Mirror Repo:
```text
Configuration/Tool: olimex-stm32-p407/kmodule,CONFIG_ARM_TOOLCHAIN_GNU_EABI
chip/stm32_gpio.c:41:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   41 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
arm-none-eabi-ld: /github/workspace/sources/nuttx/nuttx section `.bss' will not fit in region `ksram'
arm-none-eabi-ld: region `ksram' overflowed by 408 bytes
```
https://github.com/NuttX/nuttx/actions/runs/11796808033/job/32859595255#step:7:848

Which is probably caused by one of these [recent commits](https://github.com/NuttX/nuttx/commits/c1a92775a1bab7d82e4031ee277618c518c94d4c/):
- [modlib:need add ctors to init_array and dtors to fini_array](https://github.com/NuttX/nuttx/commit/c62b35a12c13b71316422c1eaca92d9a9cfd8fb2)
- [arch/arm64: bug fix,arm64_fatal_handler need regs parms](https://github.com/NuttX/nuttx/commit/d8bdf23e8a2bd929e8864b350f33d7482eea123b)
- [Remove "dumprun" functions from various LCD drivers](https://github.com/NuttX/nuttx/commit/7ebe59022c745e5ec65a421ccc1545b6a72cfcd6)
- [Fix typo errors in include/nuttx/clock.h](https://github.com/NuttX/nuttx/commit/a72f1820af4b8d287d80d483018746c6a565dbc9)
- [lib_pathbuffer:change ffs to ffsl](https://github.com/NuttX/nuttx/commit/2b190de9d14555c3335cd2062350d68a9d07eaf7)
- [fix compile warning](https://github.com/NuttX/nuttx/commit/50e89a54cb8c3180a25d42ea2429ed7cf87384a6)
- [Use lib_get_pathbuffer instead of stack variables](https://github.com/NuttX/nuttx/commit/2cf26036a5d7ec995c2edf04aa94cf07e9870883)

[(The earlier commits were OK)](https://github.com/NuttX/nuttx/commits/d502a2c16417522fcd64692631e51afd392beb7d/)

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu LTS at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Board support]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-11-12:
@lupyuen should we increase the ksram area and reduce the userspace ram?

lupyuen, 2024-11-12:
@acassis Sorry I'm not sure what's the impact?

acassis, 2024-11-12:
@lupyuen please take a look: https://github.com/apache/nuttx/pull/14748

closed 2024-11-12 by xiaoxiang781216