apache/nuttx #14773: [BUG] macOS CI: `sim:libcxxtest` doesn't link with pipe2()

lupyuen, 2024-11-14:
### Description / Steps to reproduce the issue

`sim:libcxxtest` in macOS CI doesn't link with pipe2(). We may need to disable the build because NuttX Mirror Build is failing twice daily: https://github.com/apache/nuttx/pull/14772

From sim-02 macOS Log: https://github.com/NuttX/nuttx/actions/runs/11825497334/job/32949471137

```text
Configuration/Tool: sim/libcxxtest
libcxx/libcxx/src/exception.cpp:14:12: fatal error: 'cxxabi.h' file not found
  #include <cxxabi.h>
           ^~~~~~~~~~
1 error generated.
make[1]: *** [libcxx/libcxx/src/exception.o] Error 1
In file included from libcxx/libcxx/src/stdexcept.cpp:16:
/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src/support/runtime/stdexcept_default.ipp:14:10: fatal error: 'cxxabi.h' file not found
#include <cxxabi.h>
         ^~~~~~~~~~
1 error generated.
make[1]: *** [libcxx/libcxx/src/stdexcept.o] Error 1
make[1]: Target `libxx.a' not remade because of errors.
make: *** [libs/libxx/libxx.a] Error 2
./libcxx/libcxx/src/exception.cpp:14:12: fatal error: 'cxxabi.h' file not found
  #include <cxxabi.h>
           ^~~~~~~~~~
1 error generated.
ERROR: c++ failed: 1
       command: c++ -MT ./libcxx/libcxx/src/exception.o  -M '-g' '-fomit-frame-pointer' '-fno-common' '-fvisibility=hidden' '-ffunction-sections' '-fdata-sections' '-Wall' '-Wshadow' '-Wundef' '-Wno-attributes' '-Wno-unknown-pragmas' '-nostdinc++' '-std=gnu++20' '-fno-pic' '-mcmodel=large' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include/libcxx' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include' '-D__NuttX__' '-U_AIX' '-U_WIN32' '-U__APPLE__' '-U__FreeBSD__' '-U__NetBSD__' '-U__linux__' '-U__sun__' '-U__unix__' '-U__ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__' '-D__KERNEL__' '-Wno-cpp' '-Werror' '-DLIBCXX_BUILDING_LIBCXXABI' '-D_LIBCPP_DISABLE_AVAILABILITY' '-D_LIBCPP_BUILDING_LIBRARY' '-I' '/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src' '-D__GLIBCXX__' '-Wno-shadow' '-Wno-sign-compare' '-Wno-cpp' '-Wno-attributes' '-Wno-deprecated-declarations' ./libcxx/libcxx/src/exception.cpp
make[2]: *** [libcxx/libcxx/src/exception.ddp] Error 1
In file included from ./libcxx/libcxx/src/stdexcept.cpp:16:
/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src/support/runtime/stdexcept_default.ipp:14:10: fatal error: 'cxxabi.h' file not found
#include <cxxabi.h>
         ^~~~~~~~~~
1 error generated.
ERROR: c++ failed: 1
       command: c++ -MT ./libcxx/libcxx/src/stdexcept.o  -M '-g' '-fomit-frame-pointer' '-fno-common' '-fvisibility=hidden' '-ffunction-sections' '-fdata-sections' '-Wall' '-Wshadow' '-Wundef' '-Wno-attributes' '-Wno-unknown-pragmas' '-nostdinc++' '-std=gnu++20' '-fno-pic' '-mcmodel=large' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include/libcxx' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include' '-D__NuttX__' '-U_AIX' '-U_WIN32' '-U__APPLE__' '-U__FreeBSD__' '-U__NetBSD__' '-U__linux__' '-U__sun__' '-U__unix__' '-U__ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__' '-D__KERNEL__' '-Wno-cpp' '-Werror' '-DLIBCXX_BUILDING_LIBCXXABI' '-D_LIBCPP_DISABLE_AVAILABILITY' '-D_LIBCPP_BUILDING_LIBRARY' '-I' '/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src' '-D__GLIBCXX__' '-Wno-shadow' '-Wno-sign-compare' '-Wno-cpp' '-Wno-attributes' '-Wno-deprecated-declarations' ./libcxx/libcxx/src/stdexcept.cpp
make[2]: *** [libcxx/libcxx/src/stdexcept.ddp] Error 1
make[2]: Target `makedepfile' not remade because of errors.
make[1]: *** [.depend] Error 2
make[1]: Target `depend' not remade because of errors.
make: *** [pass2dep] Error 2
make: Target `all' not remade because of errors.
/Users/runner/work/nuttx/nuttx/sources/nuttx/tools/testbuild.sh: line 385: /Users/runner/work/nuttx/nuttx/sources/nuttx/../nuttx/nuttx.manifest: No such file or directory
```

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS x64 at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-11-14:
please try CONFIG_PIPES=y, @lupyuen

lupyuen, 2024-11-14:
@xiaoxiang781216 Yep it compiles OK on macOS Arm64 yay! But wonder if it's supposed to Seg Fault?

https://gist.github.com/lupyuen/a4a6351912ebb49fc8d9db37667465e1
```text
$ tools/configure.sh sim:lua
$ kconfig-tweak --enable CONFIG_PIPES
$ grep CONFIG_PIPES .config
CONFIG_PIPES=y
$ make olddefconfig
$ make
LD:  nuttx
Pac SIM with dynamic libs..
/bin/bash: ldd: command not found
/bin/bash: readelf: command not found
SIM elf with dynamic libs archive in nuttx.tgz

$ ./nuttx
[1]    79532 segmentation fault  ./nuttx
```

simbit18, 2024-11-14:
macOS also doesn't like this

```
Pac SIM with dynamic libs..
/bin/bash: ldd: command not found
/bin/bash: readelf: command not found
SIM elf with dynamic libs archive in nuttx.tgz

```

define POSTBUILD
	$(Q)echo "Pac SIM with dynamic libs..";
	@ rm -rf sim-pac;
	@ mkdir -p sim-pac/libs;
	@ cp nuttx sim-pac/nuttx;
	@ ldd sim-pac/nuttx | grep "=> /" | awk '{print $$3}' | xargs -I '{}' cp -v '{}' sim-pac/libs;
	@ readelf -l nuttx | grep "program interpreter" | awk -F':' '{print $$2}'| cut -d"]" -f1 | xargs -I '{}' cp -v '{}' sim-pac;
	@ cp $(TOPDIR)/tools/simlaunch.sh sim-pac;
	@ tar -czf nuttx.tgz sim-pac;
	$(Q)echo "SIM elf with dynamic libs archive in nuttx.tgz"
	@ rm -rf sim-pac;
endef

xiaoxiang781216, 2024-11-14:
@xuxin930 please skip pack on macOS.

xuxin930, 2024-11-15:
> macOS also doesn't like this

hi @simbit18 could you please try this https://github.com/apache/nuttx/pull/14800 ?

simbit18, 2024-11-15:
hi @xuxin930 the pack skip is ok.

xiaoxiang781216, 2024-12-04:
@lupyuen was this issue fixed?

lupyuen, 2024-12-04:
@xiaoxiang781216 `sim:lua` is still failing for macOS (sim-02):

https://github.com/NuttX/nuttx/actions/runs/12162181362/job/33918634071#step:7:15665

```text
Configuration/Tool: sim/lua
ld: Undefined symbols:
  _pipe2, referenced from:
      _main in nuttx.rel
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

yamt, 2024-12-17:
i will take a look.

yamt, 2024-12-17:
this was not a macOS issue, but libuv's.

yamt, 2024-12-17:
> @xiaoxiang781216 Yep it compiles OK on macOS Arm64 yay! But wonder if it's supposed to Seg Fault?

i will file a separate issue for that.

yamt, 2024-12-17:
> > @xiaoxiang781216 Yep it compiles OK on macOS Arm64 yay! But wonder if it's supposed to Seg Fault?
> 
> i will file a separate issue for that.

https://github.com/apache/nuttx/issues/15228