apache/nuttx #14774: [BUG] macOS CI: `sim:libcxxtest` can't find cxxabi.h

lupyuen, 2024-11-14:
### Description / Steps to reproduce the issue

`sim:libcxxtest` build in macOS CI can't find cxxabi.h. We may need to disable the build because NuttX Mirror Build is failing twice daily: https://github.com/apache/nuttx/pull/14772

From sim-02 macOS Log: https://github.com/NuttX/nuttx/actions/runs/11825497334/job/32949471137

```text
Configuration/Tool: sim/libcxxtest
libcxx/libcxx/src/exception.cpp:14:12: fatal error: 'cxxabi.h' file not found
  #include <cxxabi.h>
           ^~~~~~~~~~
1 error generated.
make[1]: *** [libcxx/libcxx/src/exception.o] Error 1
In file included from libcxx/libcxx/src/stdexcept.cpp:16:
/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src/support/runtime/stdexcept_default.ipp:14:10: fatal error: 'cxxabi.h' file not found
#include <cxxabi.h>
         ^~~~~~~~~~
1 error generated.
make[1]: *** [libcxx/libcxx/src/stdexcept.o] Error 1
make[1]: Target `libxx.a' not remade because of errors.
make: *** [libs/libxx/libxx.a] Error 2
./libcxx/libcxx/src/exception.cpp:14:12: fatal error: 'cxxabi.h' file not found
  #include <cxxabi.h>
           ^~~~~~~~~~
1 error generated.
ERROR: c++ failed: 1
       command: c++ -MT ./libcxx/libcxx/src/exception.o  -M '-g' '-fomit-frame-pointer' '-fno-common' '-fvisibility=hidden' '-ffunction-sections' '-fdata-sections' '-Wall' '-Wshadow' '-Wundef' '-Wno-attributes' '-Wno-unknown-pragmas' '-nostdinc++' '-std=gnu++20' '-fno-pic' '-mcmodel=large' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include/libcxx' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include' '-D__NuttX__' '-U_AIX' '-U_WIN32' '-U__APPLE__' '-U__FreeBSD__' '-U__NetBSD__' '-U__linux__' '-U__sun__' '-U__unix__' '-U__ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__' '-D__KERNEL__' '-Wno-cpp' '-Werror' '-DLIBCXX_BUILDING_LIBCXXABI' '-D_LIBCPP_DISABLE_AVAILABILITY' '-D_LIBCPP_BUILDING_LIBRARY' '-I' '/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src' '-D__GLIBCXX__' '-Wno-shadow' '-Wno-sign-compare' '-Wno-cpp' '-Wno-attributes' '-Wno-deprecated-declarations' ./libcxx/libcxx/src/exception.cpp
make[2]: *** [libcxx/libcxx/src/exception.ddp] Error 1
In file included from ./libcxx/libcxx/src/stdexcept.cpp:16:
/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src/support/runtime/stdexcept_default.ipp:14:10: fatal error: 'cxxabi.h' file not found
#include <cxxabi.h>
         ^~~~~~~~~~
1 error generated.
ERROR: c++ failed: 1
       command: c++ -MT ./libcxx/libcxx/src/stdexcept.o  -M '-g' '-fomit-frame-pointer' '-fno-common' '-fvisibility=hidden' '-ffunction-sections' '-fdata-sections' '-Wall' '-Wshadow' '-Wundef' '-Wno-attributes' '-Wno-unknown-pragmas' '-nostdinc++' '-std=gnu++20' '-fno-pic' '-mcmodel=large' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include/libcxx' '-isystem' '/Users/runner/work/nuttx/nuttx/sources/nuttx/include' '-D__NuttX__' '-U_AIX' '-U_WIN32' '-U__APPLE__' '-U__FreeBSD__' '-U__NetBSD__' '-U__linux__' '-U__sun__' '-U__unix__' '-U__ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__' '-D__KERNEL__' '-Wno-cpp' '-Werror' '-DLIBCXX_BUILDING_LIBCXXABI' '-D_LIBCPP_DISABLE_AVAILABILITY' '-D_LIBCPP_BUILDING_LIBRARY' '-I' '/Users/runner/work/nuttx/nuttx/sources/nuttx/libs/libxx/libcxx/libcxx/src' '-D__GLIBCXX__' '-Wno-shadow' '-Wno-sign-compare' '-Wno-cpp' '-Wno-attributes' '-Wno-deprecated-declarations' ./libcxx/libcxx/src/stdexcept.cpp
make[2]: *** [libcxx/libcxx/src/stdexcept.ddp] Error 1
make[2]: Target `makedepfile' not remade because of errors.
make[1]: *** [.depend] Error 2
make[1]: Target `depend' not remade because of errors.
make: *** [pass2dep] Error 2
make: Target `all' not remade because of errors.
/Users/runner/work/nuttx/nuttx/sources/nuttx/tools/testbuild.sh: line 385: /Users/runner/work/nuttx/nuttx/sources/nuttx/../nuttx/nuttx.manifest: No such file or directory
```

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS x64 at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-11-14:
look like libcxxabi/Make.defs forget to add the search path.

cuiziwei1, 2024-12-05:
https://github.com/apache/nuttx/blob/3e3701b2721c216cea47b72835a13966da52b555/libs/libxx/libcxxabi/Make.defs
libcxxabi/Make.defs has already add include search path at line 38.
if enable CONFIG_LIBCXXABI, after build, you can look up whether cxxabi.h already in nuttx/include/libcxxabi.
but libcxxabi/CMakeList.txt have't add search path, I will add it.

in other way, if enable CONFIG_LIBSUPCXX_TOOLCHAIN, cxxabi.h may not be found, but if use CONFIG_LIBSUPCXX_TOOLCHAIN, we will disable LIBCXX_BUILDING_LIBCXXABI, so we can't use cxxabi.h.like this:

#if defined(LIBCXXRT) || defined(LIBCXX_BUILDING_LIBCXXABI)
  #include <cxxabi.h>
  using namespace __cxxabiv1;
  #define HAVE_DEPENDENT_EH_ABI 1
#endif

closed 2024-12-16 by xiaoxiang781216

xiaoxiang781216, 2025-02-05:
> https://github.com/apache/nuttx/blob/3e3701b2721c216cea47b72835a13966da52b555/libs/libxx/libcxxabi/Make.defs libcxxabi/Make.defs has already add include search path at line 38. if enable [...]

please provide the patch.

xiaoxiang781216, 2025-02-05:
@lupyuen do we have enough budget to bring back msvc and all macOS to ci?

lupyuen, 2025-02-05:
@xiaoxiang781216 MSVC is already in, see below. Sorry we can't enable macOS, it will exceed our budget. Remember that 1 macOS Runner = 10 times the cost of a Linux Runner.

https://github.com/apache/nuttx/actions/runs/13151470353

[image]

xiaoxiang781216, 2025-02-05:
so, we don't have any ci on macOS now:(.