apache/nuttx #14775: [BUG] macOS CI: `sim:rpproxy_virtio` fails at uintmax_t: 'unsigned long long' vs 'unsigned long'

lupyuen, 2024-11-14:
### Description / Steps to reproduce the issue

`sim:rpproxy_virtio` in macOS CI fails at uintmax_t: 'unsigned long long' vs 'unsigned long'. The same problem happens with `sim:rpserver_virtio`. We may need to disable the builds because NuttX Mirror Build is failing twice daily: https://github.com/apache/nuttx/pull/14772

From sim-03 macOS Log: https://github.com/NuttX/nuttx/actions/runs/11825497334/job/32949471805
```text
Configuration/Tool: sim/rpproxy_virtio
##[error]syslog/vsyslog.c:214:32: error: format specifies type 'uintmax_t' (aka 'unsigned long long') but the argument has type 'uintmax_t' (aka 'unsigned long') [-Werror,-Wformat]
                             , (uintmax_t)ts.tv_sec
                               ^~~~~~~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [vsyslog.o] Error 1
##[error]mm_heap/mm_memalign.c:292:9: error: format specifies type 'unsigned int' but the argument has type 'uintptr_t' (aka 'unsigned long') [-Werror,-Wformat]
        rawchunk, alignedchunk, size);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/debug.h:166:23: note: expanded from macro 'minfo'
#  define minfo       _none
                      ^
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/debug.h:108:43: note: expanded from macro '_none'
    do { if (0) syslog(LOG_ERR, format, ##__VA_ARGS__); } while (0)
                                ~~~~~~    ^~~~~~~~~~~
##[error]mm_heap/mm_memalign.c:292:19: error: format specifies type 'unsigned int' but the argument has type 'uintptr_t' (aka 'unsigned long') [-Werror,-Wformat]
        rawchunk, alignedchunk, size);
        ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/debug.h:166:23: note: expanded from macro 'minfo'
#  define minfo       _none
                      ^
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/debug.h:108:43: note: expanded from macro '_none'
    do { if (0) syslog(LOG_ERR, format, ##__VA_ARGS__); } while (0)
                                ~~~~~~    ^~~~~~~~~~~
2 errors generated.
make[1]: *** [bin/mm_memalign.o] Error 1
make[1]: Target `libdrivers.a' not remade because of errors.
make: *** [drivers/libdrivers.a] Error 2
##[error]inttypes/lib_strtoimax.c:100:22: error: implicit conversion from 'long long' to 'intmax_t' (aka 'long') changes value from -9223372036854775808 to 0 [-Werror,-Wconstant-conversion]
              return INTMAX_MIN;
              ~~~~~~ ^~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:136:29: note: expanded from macro 'INTMAX_MIN'
#  define INTMAX_MIN        INT64_MIN
                            ^~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:65:41: note: expanded from macro 'INT64_MIN'
#  define INT64_MIN         (-INT64_MAX - 1)
                             ~~~~~~~~~~~^~~
##[error]inttypes/lib_strtoimax.c:103:37: error: implicit conversion from 'long long' to 'intmax_t' (aka 'long') changes value from -9223372036854775808 to 0 [-Werror,-Wconstant-conversion]
          return (accum == limit) ? INTMAX_MIN : -(intmax_t)accum;
          ~~~~~~                    ^~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:136:29: note: expanded from macro 'INTMAX_MIN'
#  define INTMAX_MIN        INT64_MIN
                            ^~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:65:41: note: expanded from macro 'INT64_MIN'
#  define INT64_MIN         (-INT64_MAX - 1)
                             ~~~~~~~~~~~^~~
##[error]inttypes/lib_strtoimax.c:106:17: error: result of comparison of constant 9223372036854775807 with expression of type 'uintmax_t' (aka 'unsigned long') is always false [-Werror,-Wtautological-constant-out-of-range-compare]
      if (accum > INTMAX_MAX)
          ~~~~~ ^ ~~~~~~~~~~
##[error]inttypes/lib_strtoimax.c:109:18: error: implicit conversion from 'long long' to 'intmax_t' (aka 'long') changes value from 9223372036854775807 to -1 [-Werror,-Wconstant-conversion]
          return INTMAX_MAX;
          ~~~~~~ ^~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:137:29: note: expanded from macro 'INTMAX_MAX'
#  define INTMAX_MAX        INT64_MAX
                            ^~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:66:29: note: expanded from macro 'INT64_MAX'
#  define INT64_MAX         INT64_C(9223372036854775807)
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/arch/inttypes.h:35:23: note: expanded from macro 'INT64_C'
#  define INT64_C(x)  x ## ll
                      ^~~~~~~
<scratch space>:11:1: note: expanded from here
9223372036854775807ll
^~~~~~~~~~~~~~~~~~~~~
4 errors generated.
make[1]: *** [bin/lib_strtoimax.o] Error 1
##[error]inttypes/lib_strtoumax.c:91:23: error: implicit conversion from 'unsigned long long' to 'uintmax_t' (aka 'unsigned long') changes value from 18446744073709551615 to 4294967295 [-Werror,-Wconstant-conversion]
              accum = UINTMAX_MAX;
                    ~ ^~~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:140:29: note: expanded from macro 'UINTMAX_MAX'
#  define UINTMAX_MAX       UINT64_MAX
                            ^~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/stdint.h:67:29: note: expanded from macro 'UINT64_MAX'
#  define UINT64_MAX        UINT64_C(18446744073709551615)
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/runner/work/nuttx/nuttx/sources/nuttx/include/arch/inttypes.h:36:23: note: expanded from macro 'UINT64_C'
#  define UINT64_C(x) x ## ull
                      ^~~~~~~~
<scratch space>:7:1: note: expanded from here
18446744073709551615ull
^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [bin/lib_strtoumax.o] Error 1
make[1]: Target `libmm.a' not remade because of errors.
make: *** [mm/libmm.a] Error 2
##[error]iperf.c:325:14: error: format specifies type 'uintmax_t' (aka 'unsigned long long') but the argument has type 'uintmax_t' (aka 'unsigned long') [-Werror,-Wformat]
             now_len -last_len,
             ^~~~~~~~~~~~~~~~~
##[error]iperf.c:340:14: error: format specifies type 'uintmax_t' (aka 'unsigned long long') but the argument has type 'uintmax_t' (aka 'unsigned long') [-Werror,-Wformat]
             now_len,
             ^~~~~~~
2 errors generated.
make[2]: *** [iperf.c.Users.runner.work.nuttx.nuttx.sources.apps.netutils.iperf.o] Error 1
make[2]: Target `all' not remade because of errors.
make[1]: *** [/Users/runner/work/nuttx/nuttx/sources/apps/netutils/iperf_all] Error 2
##[error]nsh_dbgcmds.c:444:28: error: format specifies type 'uintmax_t' (aka 'unsigned long long') but the argument has type 'uintmax_t' (aka 'unsigned long') [-Werror,-Wformat]
                           (uintmax_t)skip);
                           ^~~~~~~~~~~~~~~
##[error]nsh_dbgcmds.c:473:20: error: format specifies type 'uintmax_t' (aka 'unsigned long long') but the argument has type 'uintmax_t' (aka 'unsigned long') [-Werror,-Wformat]
                   (uintmax_t)position);
                   ^~~~~~~~~~~~~~~~~~~
2 errors generated.
make[2]: *** [nsh_dbgcmds.c.Users.runner.work.nuttx.nuttx.sources.apps.nshlib.o] Error 1
make[2]: Target `all' not remade because of errors.
make[1]: *** [/Users/runner/work/nuttx/nuttx/sources/apps/nshlib_all] Error 2
make[1]: Target `all' not remade because of errors.
make: *** [/Users/runner/work/nuttx/nuttx/sources/apps/libapps.a] Error 2
##[error]time/lib_strftime.c:584:52: error: format specifies type 'uintmax_t' (aka 'unsigned long long') but the argument has type 'uintmax_t' (aka 'unsigned long') [-Werror,-Wformat]
               len = snprintf(dest, chleft, "%ju", (uintmax_t)mktime(&tmp));
                                             ~~~   ^~~~~~~~~~~~~~~~~~~~~~~
                                             %ju
1 error generated.
make[1]: *** [bin/lib_strftime.o] Error 1
make[1]: Target `libc.a' not remade because of errors.
make: *** [libs/libc/libc.a] Error 2
make: Target `all' not remade because of errors.
```

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS x64 at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

wyr-7, 2024-11-14:
As we see in https://github.com/apache/nuttx/blob/master/tools/ci/testlist/sim-03.dat#L3C1-L8C1,  macOS doesn't support 32bit anymore in sim:rpproxy and sim::rpserver which is CONFIG_SIM_M32=y. Consequently, it also does not support sim:rpproxy_virtio and sim:rpserver_virtio, should we also disable these?

wyr-7, 2024-11-15:
Since macOS doesn't support 32bit anymore, we change sim:rpproxy_virtio and sim:rpserver_virtio to 64bit in https://github.com/apache/nuttx/pull/14797

closed 2024-11-15 by xiaoxiang781216