apache/nuttx #14777: [BUG] Networking `HAVE_PFINET_SOCKETS` always undefined

liamHowatt, 2024-11-14:
### Description / Steps to reproduce the issue

Defconfig `sim:userfs`

```
nsh> userfs
ERROR: userfs_run() returned: -97
```

`Address family not supported by protocol` error from `net_sockif` of [net_sockif.c](https://github.com/apache/nuttx/blob/d4f6cc229d55a0c252ba6a821b9f1789e53d4bbc/net/socket/net_sockif.c#L80-L83) due to `HAVE_PFINET_SOCKETS` being wrongly undefined. The necessary `NET_UDP_HAVE_STACK` and `NET_TCP_HAVE_STACK` are not being included by [netconfig.h](https://github.com/apache/nuttx/blob/d4f6cc229d55a0c252ba6a821b9f1789e53d4bbc/include/nuttx/net/netconfig.h#L78-L81). `NET_UDP_HAVE_STACK` and `NET_TCP_HAVE_STACK` are defined in [nuttx/net/udp/udp.h](https://github.com/apache/nuttx/blob/d4f6cc229d55a0c252ba6a821b9f1789e53d4bbc/net/udp/udp.h#L52) and [nuttx/net/tcp/tcp.h](https://github.com/apache/nuttx/blob/d4f6cc229d55a0c252ba6a821b9f1789e53d4bbc/net/tcp/tcp.h#L56).

A chain of partial fixes to related to very similar issues ended with this PR #14650

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 20.04

### NuttX Version

master

### Issue Architecture

[Arch: x86_64]

### Issue Area

[Area: Networking]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-11-14:
@PetervdPerk-NXP could you look at this issue?

wengzhe, 2024-11-21:
Hi @liamHowatt , we've also encountered this bug and tried to post a fix, would you like to have a try? https://github.com/apache/nuttx/pull/14886

liamHowatt, 2024-11-21:
Thanks for addressing it @wengzhe :pray: I will take a look in ~20 hours and reply in your PR.

closed 2024-11-21 by xiaoxiang781216

liamHowatt, 2024-11-22:
Glad to see it was merged. Working again in master. Thank you. The solution makes sense to me.