apache/nuttx #14808: [BUG] rv-virt/citest: test_hello or test_pipe failed

lupyuen, 2024-11-15:
### Description / Steps to reproduce the issue

Since yesterday: `rv-virt/citest` has been failing `test_hello` onwards, or `test_pipe` onwards, hanging our CI Checks in GitHub and Build Farm. It might have been caused by one of these [NuttX Commits](https://github.com/NuttX/nuttx/commits/8ac11439c0208e2751a630461e3fc5e3a841241e/):

- [arch/toochain: Add toochain to gcc](https://github.com/NuttX/nuttx/commit/656883fec5561ca91502a26bf018473ca0229aa4)
- [mps3-an547:ap: Disable cmd_wait() as it depends on procfs currently](https://github.com/NuttX/nuttx/commit/85eed31443e60ecaf760d1212a5ec13991755585)
- [Earlier commits were OK](https://github.com/NuttX/nuttx/commits/ff4ab61b64573a97c8f24b421fcf51bb47b272a9/)

Or maybe one of these [NuttX Apps Commits](https://github.com/apache/nuttx-apps/commits/3c4ddd2802a189fccc802230ab946d50a97cb93c/):

- [nshlib/pipeline: Concat variable arguments failed](https://github.com/apache/nuttx-apps/commit/3c4ddd2802a189fccc802230ab946d50a97cb93c)
- [nshlib/nsh_parse: Fix variable arguments concat error of nsh_execute()](https://github.com/apache/nuttx-apps/commit/3a6ecb82b5f1ee670cd3bca505256f25d339e7d5)
- [Earlier commits were OK](https://github.com/nuttxpr/nuttx-apps/commits/1c7a7f7529475b0d535e2088a9c4e1532c487156/)

Also when one test fails: Why do the rest of the tests take a loooong time to fail, hanging our CI Checks in GitHub and Build Farm?

Fail at test_hello onwards: https://github.com/NuttX/nuttx/actions/runs/11833005280/job/32970891697#step:7:143
```text
Configuration/Tool: rv-virt/citest
test_framework/test_cmocka.py::test_cmocka PASSED                        [  0%]
test_example/test_example.py::test_hello FAILED                          [  0%]
test_example/test_example.py::test_helloxx FAILED                        [  0%]
test_example/test_example.py::test_pipe FAILED                           [  0%]
test_example/test_example.py::test_popen FAILED                          [  0%]
test_example/test_example.py::test_usrsocktest FAILED                    [  0%]
```

Fail at test_pipe onwards: https://github.com/NuttX/nuttx/actions/runs/11850442831/job/33025374105#step:7:145
```text
test_framework/test_cmocka.py::test_cmocka PASSED                        [  0%]
test_example/test_example.py::test_hello PASSED                          [  0%]
test_example/test_example.py::test_helloxx PASSED                        [  0%]
test_example/test_example.py::test_pipe FAILED                           [  0%]
test_example/test_example.py::test_popen FAILED                          [  0%]
test_example/test_example.py::test_usrsocktest FAILED                    [  0%]
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu LTS at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

lupyuen, 2024-11-16:
The Timeout Values are configured to One Minute or longer for some Python Tests. What if we reduce the Timeout Values? https://github.com/search?q=repo%3Aapache%2Fnuttx+timeout%3D+language%3APython+path%3A%2F%5Etools%5C%2Fci%5C%2Ftestrun%5C%2F%2F&type=code

lupyuen, 2024-11-16:
For now we patched the NuttX Mirror Repo: Kill the CI Test if it exceeds 2 hours. Also for Ubuntu Build Farm and macOS Build Farm.
- https://github.com/apache/nuttx/issues/14680#issuecomment-2480611054

xiaoxiang781216, 2024-12-04:
@lupyuen is it still broken after https://github.com/apache/nuttx/pull/14950

lupyuen, 2024-12-04:
@xiaoxiang781216 Yep CI Test is still failing, according to NuttX Dashboard:

https://nuttx-dashboard.org/d/fe2q876wubc3kc/nuttx-build-history?from=now-7d&to=now&timezone=browser&var-arch=$__all&var-subarch=$__all&var-board=rv-virt&var-config=citest&var-group=$__all&var-Filters=

[image]

https://github.com/NuttX/nuttx/actions/runs/12156309894/job/33899821697#step:7:88

```text
test_framework/test_cmocka.py::test_cmocka PASSED                        [  0%]
test_example/test_example.py::test_hello PASSED                          [  0%]
test_example/test_example.py::test_helloxx FAILED                        [  0%]
test_example/test_example.py::test_pipe FAILED                           [  0%]
test_example/test_example.py::test_popen FAILED                          [  0%]
```

lupyuen, 2024-12-11:
NuttX `ps` is crashing inside QEMU RISC-V 32-bit, causing CI Test to fail. But why is `ps` crashing for `rv-virt:citest`, but not other configs? 🤔

```bash
## Start Docker Container for NuttX
sudo docker run \
  -it \
  ghcr.io/apache/nuttx/apache-nuttx-ci-linux:latest \
  /bin/bash

## Inside Docker:
## We compile rv-virt:citest
cd
git clone https://github.com/apache/nuttx
git clone https://github.com/apache/nuttx-apps apps
pushd nuttx ; echo NuttX Source: https://github.com/apache/nuttx/tree/$(git rev-parse HEAD) ; popd
pushd apps  ; echo NuttX Apps: https://github.com/apache/nuttx-apps/tree/$(git rev-parse HEAD) ; popd
cd nuttx
tools/configure.sh rv-virt:citest
make -j
qemu-system-riscv32 \
    -M virt \
    -bios ./nuttx \
    -nographic

NuttShell (NSH) NuttX-12.7.0
nsh> uname -a
NuttX  12.7.0 5607eece84 Dec 11 2024 07:05:48 risc-v rv-virt

nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK    USED FILLED COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 0001952 0000908  46.5%  Idle_Task
    1     0 224 RR       Kthread   - Waiting  Semaphore 0000000000000000 0001904 0000508  26.6%  hpwork 0x8014b1e4 0x8014b210
    2     0 100 RR       Kthread   - Waiting  Semaphore 0000000000000000 0001896 0000508  26.7%  lpwork 0x8014b1a0 0x8014b1cc
riscv_exception: EXCEPTION: Load access fault. MCAUSE: 00000005, EPC: 80008bfe, MTVAL: 01473e00
riscv_exception: PANIC!!! Exception = 00000005
dump_assert_info: Current Version: NuttX  12.7.0 5607eece84 Dec 11 2024 07:05:48 risc-v
dump_assert_info: Assertion failed panic: at file: common/riscv_exception.c:131 task: nsh_main process: nsh_main 0x8000a806
up_dump_register: EPC: 80008bfe
```

[(See the __Complete Log__)](https://gist.github.com/lupyuen/4ec0df33c2b4b569c010fade5f471940)

tmedicci, 2024-12-12:
> NuttX `ps` is crashing inside QEMU RISC-V 32-bit, causing CI Test to fail. But why is `ps` crashing for `rv-virt:citest`, but not other configs? 🤔
> 
> ```shell
> ## Start Docker Container for NuttX
> sudo docker run \
>   -it \
>   ghcr.io/apache/nuttx/apache-nuttx-ci-linux:latest \
>   /bin/bash
> 
> ## Inside Docker:
> ## We compile rv-virt:citest
> cd
> git clone https://github.com/apache/nuttx
> git clone https://github.com/apache/nuttx-apps apps
> pushd nuttx ; echo NuttX Source: https://github.com/apache/nuttx/tree/$(git rev-parse HEAD) ; popd
> pushd apps  ; echo NuttX Apps: https://github.com/apache/nuttx-apps/tree/$(git rev-parse HEAD) ; popd
> cd nuttx
> tools/configure.sh rv-virt:citest
> make -j
> qemu-system-riscv32 \
>     -M virt \
>     -bios ./nuttx \
>     -nographic
> 
> NuttShell (NSH) NuttX-12.7.0
> nsh> uname -a
> NuttX  12.7.0 5607eece84 Dec 11 2024 07:05:48 risc-v rv-virt
> 
> nsh> ps
>   PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK    USED FILLED COMMAND
>     0     0   0 FIFO     Kthread   - Ready              0000000000000000 0001952 0000908  46.5%  Idle_Task
>     1     0 224 RR       Kthread   - Waiting  Semaphore 0000000000000000 0001904 0000508  26.6%  hpwork 0x8014b1e4 0x8014b210
>     2     0 100 RR       Kthread   - Waiting  Semaphore 0000000000000000 0001896 0000508  26.7%  lpwork 0x8014b1a0 0x8014b1cc
> riscv_exception: EXCEPTION: Load access fault. MCAUSE: 00000005, EPC: 80008bfe, MTVAL: 01473e00
> riscv_exception: PANIC!!! Exception = 00000005
> dump_assert_info: Current Version: NuttX  12.7.0 5607eece84 Dec 11 2024 07:05:48 risc-v
> dump_assert_info: Assertion failed panic: at file: common/riscv_exception.c:131 task: nsh_main process: nsh_main 0x8000a806
> up_dump_register: EPC: 80008bfe
> ```
> 
> [See the Complete Log](https://gist.github.com/lupyuen/4ec0df33c2b4b569c010fade5f471940)
> 
> [See the CI Test Log](https://gist.github.com/lupyuen/399d2ba7d964ba88cdbeb97f64778a0e)

Apparently, https://github.com/apache/nuttx/pull/15075 decreased the size of the available stack size and this makes `ps` fail on `rv-virt:citest`. I'm running some tests to confirm and submit a PR increasing the init task stack size.

tmedicci, 2024-12-12:
As I have said earlier at https://github.com/apache/nuttx/pull/15165, it didn't fix the problem of the failing CI, but fixed the problem related to `ps` command crashing.

I continued to investigate it and I was able to check that the commit that broke the CI was  656883fec5561ca91502a26bf018473ca0229aa4. I couldn't investigate it further, but I think this gives us a start point.

xiaoxiang781216, 2024-12-12:
https://github.com/apache/nuttx/commit/656883fec5561ca91502a26bf018473ca0229aa4 just modify arm specific code, it's very strange that this patch could break RISCV chip, @tmedicci .

tmedicci, 2024-12-12:
> [656883f](https://github.com/apache/nuttx/commit/656883fec5561ca91502a26bf018473ca0229aa4) just modify arm specific files, it's very strange that this patch could break RISCV chip, @tmedicci .

I will double-check. You can verify it with:

```
sudo docker run -it ghcr.io/apache/nuttx/apache-nuttx-ci-linux:latest /bin/bash -c "
  cd /tmp ; ls -la
  pwd ;
  git clone https://github.com/apache/nuttx nuttx;
  git clone https://github.com/apache/nuttx-apps apps ;
  pushd nuttx ;
  git checkout 656883fec5561ca91502a26bf018473ca0229aa4;
  echo NuttX Source: https://github.com/apache/nuttx/tree/\$(git rev-parse HEAD) ; popd ;
  pushd apps  ;
  git checkout 3a6ecb82b5f1ee670cd3bca505256f25d339e7d5;
  echo NuttX Apps: https://github.com/apache/nuttx-apps/tree/\$(git rev-parse HEAD) ; popd ;
  cd nuttx/tools/ci ; cat testlist/risc-v-05.dat;
  (./cibuild.sh -c -A -N -R testlist/risc-v-05.dat || echo '***** BUILD FAILED') ;
"
```

(and then substitute `git checkout 656883fec5561ca91502a26bf018473ca0229aa4` to `git checkout 656883fec5561ca91502a26bf018473ca0229aa4~1`)

tmedicci, 2024-12-12:
> [656883f](https://github.com/apache/nuttx/commit/656883fec5561ca91502a26bf018473ca0229aa4) just modify arm specific files, it's very strange that this patch could break RISCV chip, @tmedicci .

Yes, the test is not conclusive. I ran it twice and the results are different...

lupyuen, 2024-12-13:
Thank you so much Tiago! I created a new Bug Report for the Load Access Fault at ltp_interfaces_pthread_barrierattr_init_2_1:
- https://github.com/apache/nuttx/issues/15170

closed 2024-12-13 by lupyuen