apache/nuttx #14888: [BUG] tools/esp32/backtrace.gdbscript needs an update

yamt, 2024-11-21:
### Description / Steps to reproduce the issue

tools/esp32/backtrace.gdbscript needs an update after commit c99776659f35c34585e4fd03d29bceda1b4a2537.

this is mainly a todo for myself. but if someone wants to beat me, be my guest.

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 14.7

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Other]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-11-21:
@tmedicci @fdcavalcanti @eren-terzioglu please take a look

xiaoxiang781216, 2024-11-22:
maybe you can try the enhanced version here: https://github.com/apache/nuttx/tree/master/tools/gdb

yamt, 2024-12-06:
> maybe you can try the enhanced version here: https://github.com/apache/nuttx/tree/master/tools/gdb

does it have a similar backtrace functionality which can work for esp32?
(i couldn't find)

xiaoxiang781216, 2024-12-06:
yes, but it has much more functionality than backtrace.

yamt, 2024-12-06:
> yes, but it has much more functionality than backtrace.

ok. maybe i will try it sooner or later.