apache/nuttx #14936: [BUG] cmake on macOS is broken

yamt, 2024-11-25:
### Description / Steps to reproduce the issue

cmake sim build on macOS is broken.

it's partly because of https://github.com/apache/nuttx/pull/14852.
it doesn't seem like the only issue though.

```
(venv) spacetanuki% cmake -B build -DBOARD_CONFIG=sim:libcxxtest -GNinja
-- Initializing NuttX
  Select HOST_MACOS=y
  Select HOST_X86_64=y
--   CMake:  3.30.4
--   Ninja:  1.12.1
--   Board:  sim
--   Config: libcxxtest
--   Appdir: /Users/yamamoto/git/nuttx/apps
-- The C compiler identification is AppleClang 16.0.0.16000026
-- The CXX compiler identification is AppleClang 16.0.0.16000026
-- The ASM compiler identification is AppleClang
-- Found assembler: /Users/yamamoto/git/nuttx/prebuilt/ccache/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Users/yamamoto/git/nuttx/prebuilt/ccache/bin/cc
-- Check for working C compiler: /Users/yamamoto/git/nuttx/prebuilt/ccache/bin/cc - broken
CMake Error at /usr/local/Cellar/cmake/3.30.4/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Users/yamamoto/git/nuttx/prebuilt/ccache/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/yamamoto/git/nuttx/nuttx/build/CMakeFiles/CMakeScratch/TryCompile-XwhWZC'

    Run Build Command(s): /usr/local/bin/ninja -v cmTC_3fc14
    [1/2] /Users/yamamoto/git/nuttx/prebuilt/ccache/bin/cc   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.1.sdk -mmacosx-version-min=14.7 -U_AIX -U_WIN32 -U__APPLE__ -U__FreeBSD__ -U__NetBSD__ -U__linux__ -U__sun__ -U__unix__ -U__ENVIRONMENT_MAC_OS_X_VERSION_MIN_REQUIRED__ -fno-strict-aliasing -fomit-frame-pointer -fno-common -fvisibility=hidden -ffunction-sections -fdata-sections -Wall -Wshadow -Wundef -Wno-attributes -Wno-unknown-pragmas -Wstrict-prototypes -Wno-deprecated-declarations -no-pie -MD -MT CMakeFiles/cmTC_3fc14.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_3fc14.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_3fc14.dir/testCCompiler.c.o -c /Users/yamamoto/git/nuttx/nuttx/build/CMakeFiles/CMakeScratch/TryCompile-XwhWZC/testCCompiler.c
    clang: warning: argument unused during compilation: '-no-pie' [-Wunused-command-line-argument]
    [2/2] : && /Users/yamamoto/git/nuttx/prebuilt/ccache/bin/cc -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.1.sdk -mmacosx-version-min=14.7 -Wl,-search_paths_first -Wl,-headerpad_max_install_names -Wl,-no-pie -Wl,-dead_strip CMakeFiles/cmTC_3fc14.dir/testCCompiler.c.o -o cmTC_3fc14   && :
    FAILED: cmTC_3fc14
    : && /Users/yamamoto/git/nuttx/prebuilt/ccache/bin/cc -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.1.sdk -mmacosx-version-min=14.7 -Wl,-search_paths_first -Wl,-headerpad_max_install_names -Wl,-no-pie -Wl,-dead_strip CMakeFiles/cmTC_3fc14.dir/testCCompiler.c.o -o cmTC_3fc14   && :
    ld: unknown options: -no-pie
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:407 (project)

-- Configuring incomplete, errors occurred!
(venv) spacetanuki% 
```

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 14.7

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

yamt, 2024-11-25:
@cuiziwei1 can you take a look? thank you.

cuiziwei1, 2024-12-05:
sorry about this, can you check if CONFIG_HOST_MACOS is enable?

xuxin930, 2024-12-05:
Hi @yamt @cuiziwei1 
`CONFIG_HOST_MACOS ` is loaded after CMake generator execution completed.

However, this error occurs during the `ToolChain file `verification phase.
At this time, the generator has not yet been successfully executed.
So `CONFIG_HOST_MACOS` cannot be read

let change` NOT CONFIG_HOST_MACOS` to `NOT APPLE`