apache/nuttx #14952: [BUG] cmake-based build should not modify CMAKE_CURRENT_SOURCE_DIR

yamt, 2024-11-26:
### Description / Steps to reproduce the issue

it seems that cmake-based build sometimes modifies files under CMAKE_CURRENT_SOURCE_DIR.
eg. it expands littlefs source tree to fs/littlefs/littlefs.
it should be done under CMAKE_BINARY_DIR instead.
we should treat CMAKE_CURRENT_SOURCE_DIR read-only.

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 14.7

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-12-05:
@xuxin930 please look at this problem.

yamt, 2024-12-06:
a problem is that sometimes it prevents users from using multiple build dirs with a single source tree.
eg. when fetched data is configuration-dependent. (different versions, different local patches, etc)

IMO, it's simpler to put downloaded files under the build dir.

if we want to to have a shared cache of downloaded files, maybe it's cleaner to use a dedicated dir, instead of the source dir.