apache/nuttx #14987: [BUG] <title> Capture driver started to show wrong values on Xtensa chips

eren-terzioglu, 2024-11-29:
### Description / Steps to reproduce the issue

After [this PR](https://github.com/apache/nuttx/pull/14761) capture driver started to get some wrong values.

Steps to build:

`esp32s3-devkit:capture` configuration used to build with these options enabled:

```
CONFIG_DEBUG_FEATURES=y
CONFIG_DEBUG_ASSERTIONS=y
CONFIG_DEBUG_CAPTURE=y
CONFIG_DEBUG_CAPTURE_ERROR=y
CONFIG_DEBUG_CAPTURE_WARN=y
CONFIG_DEBUG_CAPTURE_INFO=y
CONFIG_ESP32S3_LEDC=y
CONFIG_ESP32S3_LEDC_TIM0=y
CONFIG_ESP32S3_LEDC_CHANNEL0_PIN=11
CONFIG_DEV_GPIO=y
CONFIG_ESP32S3_GPIO_IRQ=y
CONFIG_ESP_MCPWM_CAPTURE_CH1=y
CONFIG_ESP_MCPWM_CAPTURE_CH1_GPIO=13
CONFIG_ESP_MCPWM_CAPTURE_CH2=y
CONFIG_ESP_MCPWM_CAPTURE_CH2_GPIO=14
```

Sample application used to test it(I can share the code, if needed). Working code before than that PR started to show some unrelated values.

Here is some capture outputs, driver seems fine:

[image]
[image]

Note: If we close `CONFIG_DEBUG_CAPTURE_INFO=y` option, code seems working fine which closes to print information [here](https://github.com/apache/nuttx/blob/master/arch/xtensa/src/common/espressif/esp_mcpwm.c#L1543) and [here](https://github.com/apache/nuttx/blob/master/arch/xtensa/src/common/espressif/esp_mcpwm.c#L1571)

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 20.04

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Drivers]

### Verification

- [X] I have verified before submitting the report.

eren-terzioglu, 2024-11-29:
@yamt Please have a look

acassis, 2024-12-07:
@eren-terzioglu @yamt should we revert https://github.com/apache/nuttx/pull/14761 ?

eren-terzioglu, 2024-12-09:
@acassis Seems it is working for now. If I face issue again, I will reopen the issue. Thanks

closed 2024-12-09 by eren-terzioglu