apache/nuttx #14998: [BUG] <title>quot and rem in lldiv_t are of the wrong type

v-clone, 2024-11-29:
### Description / Steps to reproduce the issue

The 'quot' and 'rem' members of the lldiv_s in stdlib.h are of type 'long' when they should be of type 'long long'.

### On which OS does this issue occur?

[OS: Other]

### What is the version of your OS?

RHEL 7.x

### NuttX Version

master/all

### Issue Architecture

[Arch: arm], [Arch: mips]

### Issue Area

[Area: Posix]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-12-01:
@v-clone thanks for reporting this issue! Please submit a PR to fix it.

maxikrie, 2026-10-30:
@acassis Both quot and rem in struct lldiv_s are now long long in include/stdlib.h, as requested here. This appears fixed; can we close this issue?

closed 2026-10-30 by acassis