apache/nuttx #15016: [BUG] macOS CI: Error: common/riscv_exit.c:65:33: error: 'tcb' undeclared (first use in this function);

simbit18, 2024-12-02:
### Description / Steps to reproduce the issue

bl602evb/wifi in macOS CI fails 
Error: common/riscv_exit.c:65:33: error: 'tcb' undeclared (first use in this function); did you mean 'tcb_s'?
   65 |   g_running_tasks[this_cpu()] = tcb;
      |                                 ^~~
      |                                 tcb_s
common/riscv_exit.c:65:33: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [riscv_exit.o] Error 1

The same problem happens with
esp32c3-devkit/cxx
esp32c3-devkit/wifi

MacOS-13 [macos]
https://github.com/simbit18/nuttx_test_pr/actions/runs/12117323247/job/33779603978

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

MacOS 13

### NuttX Version

master

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

hujun260, 2024-12-02:
https://github.com/apache/nuttx/pull/15014

closed 2024-12-02 by xiaoxiang781216

simbit18, 2024-12-02:
test ok
https://github.com/simbit18/nuttx_test_pr/actions/runs/12118869734/job/33784342548