apache/nuttx #15020: [BUG] RTCB can't be null since g_readytorun is in bss section

PetervdPerk-NXP, 2024-12-02:
### Description / Steps to reproduce the issue

https://github.com/apache/nuttx/blob/master/sched/sched/sched_lock.c#L80-L85
Mentions that `rtcb may be NULL only during early boot-up phases` however RTCB ends up in bss.
```
20002450 g     O .bss   00000008 g_readytorun
```
Therefore rtcb is a random value instead of NULL.

Not sure what would be better remove the sched_lock check or initialize g_readytorun as 0.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu

### NuttX Version

Master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: OS Components]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-12-03:
@hujun260 @anchao please take a look!

acassis, 2024-12-03:
@PetervdPerk-NXP .bss area is suppose to be zeroed early in the boot process, so it will not be a random value as you said.

acassis, 2024-12-03:
This is from stm32_start.c:

```
  stm32_clockconfig();
  arm_fpuconfig();
  stm32_lowsetup();
  stm32_gpioinit();
  showprogress('A');

  /* Clear .bss.  We'll do this inline (vs. calling memset) just to be
   * certain that there are no issues with the state of global variables.
   */

  for (dest = (uint32_t *)_START_BSS; dest < (uint32_t *)_END_BSS; )
    {
      *dest++ = 0;
    }

  showprogress('B');
```

anchao, 2024-12-04:
It looks similar to the problem @PeterBee97 solved before https://github.com/apache/nuttx/pull/14889 .
Before BSS is initialized, the boot-up code should avoid using any kernel API.

This is a common problem, we need to adjust the startup code to ensure BSS initialization earlier. @GUIDINGLI @hujun260  could you arrange resources to invest in this? I am not working on nuttx projects recently. CC @xiaoxiang781216

xiaoxiang781216, 2024-12-04:
The problem is clear that .bss need to be initialized before os, even before any c code, except the code which zero memory.

casaroli, 2026-11-08:
Fixed by #17300.