apache/nuttx #15063: [BUG] <fatal: could not read Username for 'https://github.com': No such device or address>

cuiziwei1, 2024-12-05:
### Description / Steps to reproduce the issue

Normalize esp32-devkitc/mqttc
====================================================================================
Configuration/Tool: esp32-devkitc/nxdiag
2024-12-05 13:07:23
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Building NuttX...
fatal: could not read Username for 'https://github.com/': No such device or address
make[3]: *** [Makefile:83: espressif_prepare] Error 128
make[3]: Target 'context' not remade because of errors.
make[2]: *** [Makefile:55: /github/workspace/sources/apps/system/nxdiag_context] Error 2
make[2]: Target 'context_all' not remade because of errors.
make[1]: *** [Makefile:185: context] Error 2
make: *** [tools/Unix.mk:457: /github/workspace/sources/apps/.context] Error 2
make: Target 'all' not remade because of errors.
/github/workspace/sources/nuttx/tools/testbuild.sh: line 385: /github/workspace/sources/nuttx/../nuttx/nuttx.manifest: No such file or directory

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

ci

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Other]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-12-05:
@tmedicci please look at this problem, which happen today with every pr.

tmedicci, 2024-12-05:
I'm testing https://github.com/apache/nuttx/issues/15063#issuecomment-2520403890 to solve this issue. It didn't fail running the CI locally.

xiaoxiang781216, 2024-12-06:
could you split the fix from https://github.com/apache/nuttx-apps/pull/2879? so we can merge it as fast as possible.

closed 2024-12-06 by cuiziwei1

tmedicci, 2024-12-06:
> could you split the fix from [apache/nuttx-apps#2879](https://github.com/apache/nuttx-apps/pull/2879)? so we can merge it as fast as possible.

I will do it right now.