apache/nuttx #15065: [BUG] Crash Introduced when calling app from nsh

stbenn, 2024-12-05:
### Description / Steps to reproduce the issue

The simple Hello World example app is crashing when called from nsh on Coretex-M33 (armv8-m). It appears to be introduced in 2b22ee03. 

## Steps to Reproduce
This bug was found using the `nucleo-h563zi:nsh` configuration, and the most up to date apps commit (6600a5fd)

1. Checkout apps master (6600a5fd) and nuttx master (or 2b22ee03)
2. cd to nuttx directory
3. `make distclean`
4. `./tools/configure.sh -l nucleo-h563zi:nsh`
5. `make menuconfig`
6. Application Configuration -> Examples -> Enable "Hello, World!" example
7. Build and debug while monitoring the serial console.
8. In serial console, `nsh> hello`

### Log Files
Earliest commit it occurs: [hello_crash_log_2b22ee03d6.txt](https://github.com/user-attachments/files/18024668/hello_crash_log_2b22ee03d6.txt)
Current master branch: [hello_crash_log_75fc19d729.txt](https://github.com/user-attachments/files/18024671/hello_crash_log_75fc19d729.txt)

 

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Drivers], [Area: OS Components]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-12-05:
@hujun260 please take a look, any idea why your PR could have introduced this issue?

stbenn, 2024-12-05:
I checked with the Nucleo-G071RB and got what appears to be the same crash. Follow the same steps as described above, but with `nucleo-g071rb:nsh` configuration. Here is the [log file](https://github.com/user-attachments/files/18024990/nucleo-g071rb_crash_2b22ee03d6.txt)

stbenn, 2024-12-05:
If it helps, `PANIC()` at the end of `up_exit()` in `/nuttx/arch/arm/src/common/arm_exit.c` is reached.

hujun260, 2024-12-06:
> @hujun260 please take a look, any idea why your PR could have introduced this issue?

ok Let me analyze this issue

hujun260, 2024-12-06:
https://github.com/apache/nuttx/pull/15073

closed 2024-12-09 by xiaoxiang781216