apache/nuttx #15148: [BUG] Build error after switching from release/12.7 to master with LittleFS enabled

FelipeMdeO, 2024-12-11:
### Description / Steps to reproduce the issue

Description:
Hello everyone,
After switching from the release/12.7 branch to master, I am encountering a build error when LittleFS is enabled in the filesystem. This issue does not occur when using the release/12.7 branch.

Steps to Reproduce:

    Use the following configuration:

./tools/configure.sh esp32c6-devkitc:nsh

Enable LittleFS in the filesystem.
Build the project:

    make -j

    Enter menuconfig and activate any additional feature (e.g., hello_world).
    Build again.

Expected Behavior:
The build should complete successfully.

Actual Behavior:
The following error occurs during the build process:

felipe-moura@felipe-moura:~/nuttxspace/nuttx$ make -j
Create version.h
Downloading: littlefs/v2.5.1.tar.gz LN: platform/board to /home/felipe-moura/nuttxspace/nuttx-apps/platform/dummy
Register: nsh
Register: sh
Register: hello
Register: ostest
Register: getprime
Register: dumpstack
error: patch failed: fs/littlefs/littlefs/lfs_util.h:28
error: fs/littlefs/littlefs/lfs_util.h: patch does not apply
make[1]: *** [littlefs/Make.defs:65: .littlefsunpack] Error 1
make: *** [tools/Unix.mk:457: fs/.context] Error 2

Additional Information:

    This issue does not occur in the release/12.7 branch.
    Environment:
        Host OS: Linux (Ubuntu)
        Branch: master

defconfig:
`#
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
CONFIG_ARCH="risc-v"
CONFIG_ARCH_BOARD="esp32c6-devkitc"
CONFIG_ARCH_BOARD_COMMON=y
CONFIG_ARCH_BOARD_ESP32C6_DEVKITC=y
CONFIG_ARCH_CHIP="esp32c6"
CONFIG_ARCH_CHIP_ESP32C6=y
CONFIG_ARCH_CHIP_ESP32C6WROOM1=y
CONFIG_ARCH_INTERRUPTSTACK=2048
CONFIG_ARCH_RISCV=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_BOARDCTL_RESET=y
CONFIG_BOARD_LOOPSPERMSEC=15000
CONFIG_BUILTIN=y
CONFIG_DEV_ZERO=y
CONFIG_ESPRESSIF_ESP32C6=y
CONFIG_EXAMPLES_HELLO=y
CONFIG_FS_LITTLEFS=y
CONFIG_FS_PROCFS=y
CONFIG_IDLETHREAD_STACKSIZE=2048
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INTELHEX_BINARY=y
CONFIG_LIBC_PERROR_STDOUT=y
CONFIG_LIBC_STRERROR=y
CONFIG_NFILE_DESCRIPTORS_PER_BLOCK=6
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_READLINE=y
CONFIG_NSH_STRERROR=y
CONFIG_PREALLOC_TIMERS=0
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_BACKTRACE=y
CONFIG_SCHED_WAITPID=y
CONFIG_START_DAY=29
CONFIG_START_MONTH=11
CONFIG_START_YEAR=2019
CONFIG_SYSTEM_DUMPSTACK=y
CONFIG_SYSTEM_NSH=y
CONFIG_TESTING_GETPRIME=y
CONFIG_TESTING_OSTEST=y
CONFIG_UART0_SERIAL_CONSOLE=y
`

Please let me know if additional details or logs are needed. Thank you!

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.0

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: File System]

### Verification

- [X] I have verified before submitting the report.

yamt, 2024-12-12:
maybe this is related to my change. https://github.com/apache/nuttx/pull/14956
i will take a look later.

a question: have you stared from a clean tree?

eren-terzioglu, 2024-12-12:
Our tests seems fine on building LittleFS. Please try to build using `esp32c6-devkitc:spiflash ` defconfig with these options enabled:
```
CONFIG_ESPRESSIF_SPIFLASH_LITTLEFS=y
CONFIG_ESPRESSIF_SPIFLASH_SMARTFS=n
```

FelipeMdeO, 2024-12-12:
> maybe this is related to my change. #14956 i will take a look later.
> 
> a question: have you stared from a clean tree?

Yes, I started from clean tree.

FelipeMdeO, 2024-12-12:
> Our tests seems fine on building LittleFS. Please try to build using `esp32c6-devkitc:spiflash ` defconfig with these options:
> 
> ```
> CONFIG_ESPRESSIF_SPIFLASH_LITTLEFS=y
> CONFIG_ESPRESSIF_SPIFLASH_SMARTFS=n
> ```

Eren. I am facing issue when I try follow your steps, When I try enable ESPRESSIF_SPIFLASH_LITTLEFS the manuconfig crash, see below:
```
felipe-moura@felipe-moura:~/nuttxspace/nuttx$ make menuconfig
CP: arch/dummy/Kconfig to /home/felipe-moura/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/felipe-moura/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/felipe-moura/nuttxspace/nuttx-apps/platform/dummy
LN: include/arch to arch/risc-v/include
LN: include/arch/board to /home/felipe-moura/nuttxspace/nuttx/boards/risc-v/esp32c6/esp32c6-devkitc/include
LN: drivers/platform to /home/felipe-moura/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/felipe-moura/nuttxspace/nuttx/arch/risc-v/include/esp32c6
LN: arch/risc-v/src/chip to /home/felipe-moura/nuttxspace/nuttx/arch/risc-v/src/esp32c6
LN: arch/risc-v/src/board to /home/felipe-moura/nuttxspace/nuttx/boards/risc-v/esp32c6/esp32c6-devkitc/../common
LN: arch/risc-v/src/board/board to /home/felipe-moura/nuttxspace/nuttx/boards/risc-v/esp32c6/esp32c6-devkitc/src
Loaded configuration '.config'
Traceback (most recent call last):
  File "/home/felipe-moura/.local/bin/menuconfig", line 8, in <module>
    sys.exit(_main())
             ^^^^^^^
  File "/home/felipe-moura/.local/lib/python3.12/site-packages/menuconfig.py", line 663, in _main
    menuconfig(standard_kconfig(__doc__))
  File "/home/felipe-moura/.local/lib/python3.12/site-packages/menuconfig.py", line 732, in menuconfig
    print(curses.wrapper(_menuconfig))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/felipe-moura/.local/lib/python3.12/site-packages/menuconfig.py", line 865, in _menuconfig
    if not _change_node(sel_node):
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/felipe-moura/.local/lib/python3.12/site-packages/menuconfig.py", line 1620, in _change_node
    _leave_menu()
  File "/home/felipe-moura/.local/lib/python3.12/site-packages/menuconfig.py", line 1205, in _leave_menu
    _sel_node_i = _shown.index(_cur_menu)
                  ^^^^^^^^^^^^^^^^^^^^^^^
ValueError: <menu node for choice ESPRESSIF_SPIFLASH_FS, prompt "Mount SPI Flash MTD on bring-up" (visibility n), deps n, has help, has child, has next, boards/risc-v/esp32h2/common/Kconfig:15> is not in list
make: *** [tools/Unix.mk:701: menuconfig] Error 1
```

eren-terzioglu, 2024-12-12:
> Eren. I am facing issue when I try follow your steps, When I try enable ESPRESSIF_SPIFLASH_LITTLEFS the manuconfig crash, see below:
>
> ```
> felipe-moura@felipe-moura:~/nuttxspace/nuttx$ make [...]

Did you add options to defconfig or did you do it manually?

FelipeMdeO, 2024-12-12:
Hello @eren-terzioglu ,

I did distclean, 
Executed esp32c6-devkitc:spiflash
Opened  Menuconfig
Search for ESPRESSIF_SPIFLASH_LITTLEFS using char "/" 
Enable option. When I pressed "bar key" menuconfig crash immediately.

I am in master branch.

FelipeMdeO, 2024-12-12:
@acassis What do you thing about menuconfig crash? It is a bug also, should I create 

@eren-terzioglu  You need build 2 times to get issue, did you do it ?

eren-terzioglu, 2024-12-12:
It didn't crash on my system but choice option disappeared, probably it disables choice menu. It should be fine if you use enter key. Worst case scenario you can change file system preference under `Board Selection` menu  or using defconfig file

acassis, 2024-12-12:
@FelipeMdeO just do a test using kconfig-frontend instead of kconfig lib. I think kconfiglib is more restrictive, if it find some inconsistencies in the Kconfig file it will fail. So after confirming it works with kconfig-frontend we need to discover what changed in the Kconfig that raised this issue on kconfiglib

fdcavalcanti, 2024-12-12:
It looks like an environment issue. Try a fresh nuttx/apps clone and see if it works.

Maybe some pending symlinks for a different project?
Run a `distclean` and then `git clean -xfdn`. It will show you pending untracked files that may break your build environment.

yamt, 2024-12-13:
i couldn't reproduce the error.

i used "make -j8" instead of "make -j".
nuttx version: 4a51c21afce7f84ed28792038854d3d37b71083e

```
make distclean
./tools/configure.sh esp32c6-devkitc:nsh
make -j8
make menuconfig
make -j8
```

macOS 15.2, x86-64

FelipeMdeO, 2024-12-13:
> i couldn't reproduce the error.
> 
> i used "make -j8" instead of "make -j". nuttx version: [4a51c21](https://github.com/apache/nuttx/commit/4a51c21afce7f84ed28792038854d3d37b71083e)
> 
> ```
> make distclean
> ./tools/configure.sh esp32c6-devkitc:nsh
> make -j8
> make menuconfig
> make -j8
> ```
> 
> macOS 15.2, x86-64

Can you, please, enable littlefs after ./tools/configure.sh esp32c6-devkitc:nsh and execute other steps again?

yamt, 2024-12-16:
> Can you, please, enable littlefs after ./tools/configure.sh esp32c6-devkitc:nsh and execute other steps again?

ok.
i think i now know what caused the issue.
let me think how to fix it.

yamt, 2024-12-16:
> ok. i think i now know what caused the issue. let me think how to fix it.

https://github.com/apache/nuttx/pull/15204

closed 2024-12-16 by xiaoxiang781216