apache/nuttx #15162: [BUG] Recent modification broke raspberry pi pico nsh

acassis, 2024-12-12:
### Description / Steps to reproduce the issue

Yashvi Shah reported me that she wasn't getting the nsh working on RaspberryPi Pico, then I test here and in fact some recent modification broke the NSH.

@lupyuen is it possible to add RaspberryPi Pico to the automated test as well?

$ git clone -b 2.0.0 https://github.com/raspberrypi/pico-sdk.git
 
$ export PICO_SDK_PATH=/home/alan/pico-sdk

$ cd nuttxspace/nuttx
 
$ make distclean
 
You can use this script to find the board name after configure.sh just typing the inicial name of the board and pressing TAB
 
$ . tools/configure_completion.bash
 
$ ./tools/configure.sh raspberrypi-pico:nsh
 
$ make -j
 flash: 144 KB 2 MB 7.03%
 sram: 8224 B 264 KB 3.04%
Generating: nuttx.uf2
Done

Copy nuttx.uf2 to the virtual disk of raspberrypi pico and use a USB/Serial connected to GP0, GP1 and GND and reset the board, you will see the nsh is not accepting text and there is trash after the nuttx entry message (see attached image)
[image]

### On which OS does this issue occur?

[OS: BSD], [OS: Linux], [OS: Mac], [OS: Windows], [OS: Other]

### What is the version of your OS?

24.04

### NuttX Version

12.7 mainline

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Board support]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-12-12:
@hujun260 @xiaoxiang781216 please add RaspberryPi Pico to your automated test. Does your test verify if the board nsh> is working? As you can see in this case the nsh> was displayed but the communication is not working

acassis, 2024-12-12:
I tested "usbnsh" and it is working! So it is only related to UART... I have a suspect now! ;-)

yashvi-05, 2024-12-12:
Thank you for helping me.....

acassis, 2024-12-12:
> Thank you for helping me.....

You are welcome @yashvi-05 ! Your test help us to discover an issue caused by recent modifications.

hujun260, 2024-12-13:
> I tested "usbnsh" and it is working! So it is only related to UART... I have a suspect now! ;-)
> 
> Update: although the "nsh>" is working, I noticed the line feeding (breaking line) sometimes is failing, normally for commands like "free", "help", "uname", normally "ls" works fine.

Can you help us verify which commit caused it?

acassis, 2024-12-13:
@hujun260 sure! I will do git bisect to find the guilt commit